modm-io/modm #3: Checking out project

TotalKrill, 2018-03-20:
Hey! I was just checking this repo out, apparently a lot of the links in the readme seem down, would love to read the API documentation etc. Seems like nice work!

salkinium, 2018-03-28:
Yeah, the Readme is just copied from xpcc with some very minor polish. The C++ API is amlost identical to xpcc (except it’s now all in namespace `modm`), but obviously the generation part is missing. I’ve been busy building embedded control systems for a huge model railway at uni with modm, so didn’t get to work on the docs too much. Did write a blog post about where the data comes from tho: http://blog.salkinium.com/modm-devices/

salkinium, 2018-03-28:
I would recommend using xpcc.io for now, unless you’re here explicitly for the new HAL generator. The APIs haven’t changed much and I know from experience, that writing the docs is the hardest part, so it’ll probably take another 6 months or so to finish.

salkinium, 2018-04-21:
Hey @TotalKrill, I‘ve finally added a homepage at [modm.io](http://modm.io) that describes the project.
Only the Reference section is missing with the API, but it‘s probably best to look at the generated source code anyways. Hope this helps, tell me if anything is unclear.

TotalKrill, 2018-04-22:
Hey!

Nice! I will have a look at this a bit closer when I get the time! But just scrolling around for now, the page looks really great!

salkinium, 2018-05-23:
I'm closing this, since the homepage is now available. It may not be complete, but that is a constant fight anyways.

closed 2018-05-23 by salkinium