modm-io/modm #18: Use C++17 language features

salkinium, 2018-05-18:
We're already using C++14 by default, which requires at least GCC 5.
This is an issue for the official avr-gcc from Atmel which ships with Ubuntu in version 4.9.3.
Most other linux distributions, as well as [macOS via Homebrew](https://github.com/osx-cross/homebrew-avr) (Q: what about Windows?) compile their own avr-gcc from upstream, which is missing support for several AVR architectures and devices.
For the sake of moving forwards, we've accepted this caveat and are using a [self-compiled avr-gcc 7.3.0](https://github.com/salkinium/docker-avr-gcc-7/) for our CI.
ARM already ships arm-none-eabi-gcc for v7.2.0 since the 2017q4 release.

The [C++ standards support page for GCC](https://gcc.gnu.org/projects/cxx-status.html) shows that all language features of C++17 (except fixes to template argument deduction for class templates) are supported by GCC 7.
I would therefore like to enable C++17 as the default for modm to already make use of very useful new features such as `constexpr if`, enhancements to variadic templates, selection statements with initializers, constexpr lambdas and many more.

cc @dergraaf @strongly-typed @rleh @chris-durand @daniel-k @ekiwi @georgi-g @mhthies

salkinium, 2018-05-21:
If you use 32-bit Linux, well… meh.
I guess in the worst case you can always provide a Docker container for some precompiled avr-gcc.

I think this is happening.

rleh, 2018-05-21:
> 32-bit Linux

Who uses 32-bit Linux today?

The Ubuntu PPA and Fedora include i386/i686 builds of avr-gcc and arm-none-eabi-gcc.

Oh, Raspberry Pi still has no official 64-bit software support :unamused: 
Raspbian has no avr-gcc and arm-none-eabi-gcc packages, but Fedora has ARMv7 and Aarch64 packages of avr-gcc and arm-none-eabi-gcc.

salkinium, 2018-05-24:
> I think this is happening.

It happened.

closed 2018-05-24 by salkinium

daniel-k, 2018-05-24:
\o/

PR #21 (2018-05-24): Enable C11 and C++17 by default
Let's use some modern standards. 

- C11 requires at least GCC 4.9  (we already require GCC 5 anyways).
- C++17 requires at least GCC 7.

Closes #18. See issue for discussion on compiler availability.
cc @rleh @siy @dergraaf @strongly-typed