modm-io/modm #67: Problems using lbuild with windows

amarokmclion, 2018-08-15:
Hi all,

I'm trying to use modm under windows 10 with anaconda3 64bit. I've followed the steps in the installation page and found a few mistakes. 

- "conda install" command from anaconda can't install github recipes like "pip install" can

- pyelftools is not available from conda-forge thus I had to install it using pip

In order to solve this issues I used the following commands to procede with the installation:

```
conda create --name modm python=3 pip #this is needed because otherwise anaconda will use the global pip and not the one from the env
activate modm
conda install -c conda-forge git 
pip install jinja2 scons future pyelftools git+https://github.com/modm-io/library-builder
```
After doing this I had to configure a .bat file to run the lbuild file inside Scripts, something simple like this:

```
@echo off
set scriptname=%~dp0%~n0
python "%scriptname%" %*
```

After doing this I cloned recursively the repository and I could call lbuild inside an example directory, but I got this error:

> c:\modm\examples\stm32f4_discovery\blink (develop)
> (modm3) λ lbuild -v build
> [INFO] lbuild.module: Found module 'cmsis'
> [INFO] lbuild.module: Found module 'core'
> [INFO] lbuild.module: Found module 'dsp'
> [INFO] lbuild.module: Found module 'fatfs'
> [INFO] lbuild.module: Found module 'freertos'
> [INFO] lbuild.module: Found module 'device'
> [INFO] lbuild.module: Found module 'ros'
> [INFO] lbuild.module: Found module 'device'
> [INFO] lbuild.module: Found module 'tlsf'
> [INFO] lbuild.module: Found module 'architecture'
> [INFO] lbuild.module: Found module 'accessor'
> [INFO] lbuild.module: Found module 'adc'
> [INFO] lbuild.module: Found module 'assert'
> [INFO] lbuild.module: Found module 'atomic'
> [INFO] lbuild.module: Found module 'block.device'
> [INFO] lbuild.module: Found module 'can'
> [INFO] lbuild.module: Found module 'clock'
> [INFO] lbuild.module: Found module 'delay'
> [INFO] lbuild.module: Found module 'gpio'
> [INFO] lbuild.module: Found module 'gpio.expander'
> [INFO] lbuild.module: Found module 'heap'
> [INFO] lbuild.module: Found module 'i2c'
> [INFO] lbuild.module: Found module 'i2c.device'
> [INFO] lbuild.module: Found module 'interrupt'
> [INFO] lbuild.module: Found module 'memory'
> [INFO] lbuild.module: Found module '1-wire'
> [INFO] lbuild.module: Found module 'register'
> [INFO] lbuild.module: Found module 'spi'
> [INFO] lbuild.module: Found module 'spi.device'
> [INFO] lbuild.module: Found module 'uart'
> [INFO] lbuild.module: Found module 'unaligned'
> [INFO] lbuild.module: Found module 'board'
> [INFO] lbuild.module: Found module 'al-avreb-can'
> [INFO] lbuild.module: Found module 'arduino-uno'
> [INFO] lbuild.module: Found module 'black-pill'
> [INFO] lbuild.module: Found module 'blue-pill'
> [INFO] lbuild.module: Found module 'disco-f051r8'
> [INFO] lbuild.module: Found module 'disco-f072rb'
> [INFO] lbuild.module: Found module 'disco-f100rb'
> [INFO] lbuild.module: Found module 'disco-f303vc'
> [INFO] lbuild.module: Found module 'disco-f407vg'
> [INFO] lbuild.module: Found module 'disco-f429zi'
[... 224 lines cut ...]
> Package      Version
> ------------ ---------
> certifi      2018.8.13
> future       0.16.0
> gitdb2       2.0.4
> GitPython    2.1.11
> Jinja2       2.10
> lbuild       0.1
> lxml         4.2.4
> MarkupSafe   1.0
> pip          18.0
> pyelftools   0.24
> scons        3.0.1
> setuptools   40.0.0
> smmap2       2.0.4
> wheel        0.31.1
> wincertstore 0.2

> c:\modm\examples\stm32f4_discovery\blink (develop)
> (modm3) λ python
> Python 3.6.6 | packaged by conda-forge | (default, Jul 26 2018, 11:48:23) [MSC v.1900 64 bit (AMD64)] on win32
> Type "help", "copyright", "credits" or "license" for more information.
> >>>

I hope this will help getting a better windows support.

salkinium, 2018-08-16:
Hm, stuck on the very first access to a template… sorry about this, I haven’t used Windows in over a decade, so not an expert.

Cc @dergraaf maybe a path encoding error? I know we had path issues in SCons on Windows before.

amarokmclion, 2018-08-17:
Hi again,

I did some debugging and found that Jinja2 doesn't like the \ path separator. I'll try to craft a fix and push it in a PR.

salkinium, 2018-08-17:
In xpcc, there was a `.replace` on the template path:
https://github.com/roboterclubaachen/xpcc/blob/develop/scons/site_tools/template.py#L134
But it’s missing in lbuild:
https://github.com/modm-io/library-builder/blob/develop/lbuild/environment.py#L241

Maybe that’s the issue?

amarokmclion, 2018-08-17:
Yes, I did something like that in order to make it work. 
I've encountered also a problem with the generated SConscript, each env.File entry had a single backslash as a separator in the string and thus SCons couldn't read correctly the path string. 
I manually changed each backslash in a forward slash and it worked.
This made me remember that windows 10 supports both forward and back slash as path separators even in the command line, maybe one general solution could be to forget the backslash altogether.
The other solution is to use the paths generated with pathlib instead of os.path.

amarokmclion, 2018-08-18:
I've made some pull requests that will make life for windows users a little better.

salkinium, 2018-08-25:
I've updated the install instructions with your suggestions: https://modm.io/guide/installation/#windows

However, I'm unclear about the `.bat` step. Is this still required, or did [your lbuild PR](https://github.com/modm-io/library-builder/pull/2) fix this?

salkinium, 2018-10-18:
cc @daniel-k also has to support Windows

jnewcomb, 2018-12-01:
Hello folks,
I've failed to reach the first hurdle! Any tips / suggestions / workarounds are much appreciated.

There seems to be an issue in project.xml with the use of the : (semicolon) in   
 modm:board:nucleo-f303k8  when running lbuild build.

`(modm) C:\Users\im\modm\examples\nucleo_f303k8\blink>lbuild build
Traceback (most recent call last):
  File "c:\users\im\appdata\local\conda\conda\envs\modm\lib\runpy.py", line 193, in _run_module_as_main    "__main__", mod_spec)
  File "c:\users\im\appdata\local\conda\conda\envs\modm\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\im\AppData\Local\conda\conda\envs\modm\Scripts\lbuild.exe\__main__.py", line 9, in <module>
  File "c:\users\im\appdata\local\conda\conda\envs\modm\lib\site-packages\lbuild\main.py", line 410, in main output = run(args)
  File "c:\users\im\appdata\local\conda\conda\envs\modm\lib\site-packages\lbuild\main.py", line 395, in run builder = Builder(config=args.config, options=args.options)
  File "c:\users\im\appdata\local\conda\conda\envs\modm\lib\site-packages\lbuild\api.py", line 44, in __init__ config = ConfigNode.from_file(config)
  File "c:\users\im\appdata\local\conda\conda\envs\modm\lib\site-packages\lbuild\config.py", line 160, in from_file if cpath.exists():
  File "c:\users\im\appdata\local\conda\conda\envs\modm\lib\pathlib.py", line 1329, in exists self.stat()
  File "c:\users\im\appdata\local\conda\conda\envs\modm\lib\pathlib.py", line 1151, in stat  return self._accessor.stat(self)
OSError: [WinError 123] The filename, directory name, or volume label syntax is incorrect: 
'C:\\Users\\im\\modm\\examples\\nucleo_f303k8\\blink\\modm:board:nucleo-f303k8'`>

My setup
Windows10

'(modm) C:\Users\im\modm\examples\nucleo_f303k8\blink>pip list
Package         Version
--------------- ----------
anytree         2.4.3
certifi         2018.11.29
colorama        0.4.1
colorful        0.4.4
CppHeaderParser 2.7.4
deepdiff        3.3.0
future          0.17.1
gitdb2          2.0.5
GitPython       2.1.11
Jinja2          2.10
jsonpickle      1.0
lbuild          1.5.0
lxml            4.2.5
MarkupSafe      1.1.0
pip             18.1
ply             3.11
pyelftools      0.25
scons           3.0.1
setuptools      40.6.2
six             1.11.0
smmap2          2.0.5
wheel           0.32.3
wincertstore    0.2'

(modm) C:\Users\im\modm\examples\nucleo_f303k8\blink>python
Python 3.7.1 | packaged by conda-forge | (default, Nov 13 2018, 19:01:41) [MSC v.1900 64 bit (AMD64)] :: Anaconda, Inc. on win32

Thanks..

salkinium, 2018-12-01:
The `<extends>` element works by first checking if a file exists with this name locally, before checking in the repositories. Here is seems like pathlib doesn’t properly escape `:` in file names for Windows, when lbuild calls `cpath.exists()`. That’s pretty bad, especially since you have an up-to-date Python version. We could fix it in lbuild, but I don’t really want to add platform specific workarounds on a Python project…

salkinium, 2018-12-01:
Apparently, the symbol `:` (colon) is simply [not allowed in paths on Windows](
https://docs.microsoft.com/en-us/windows/desktop/FileIO/naming-a-file#file_and_directory_names).
I don’t understand though why this worked for some of our other Windows users and not for you, maybe this is new with Windows 10? We’ve had a lot of Windows paths issues, some with conflicting solutions, so I’m very uncertain about what the limitations of Windows paths actually are.

jnewcomb, 2018-12-02:
Time to build a build a box then..
I guess because windows maps drives using the : (colon) whereas Linux uses mounts.
So as well a / (slash) not being a valid char they have added the : (colon) too.
Never thought about that.. imagine the confusion if you called a directory c:mydir
Saying that, you call a directory  my%20files  and it doesn't seem to complain!
Thanks for a quick response,
Jon.

operativeF, 2019-02-13:
For those on Windows looking for a quick fix to this, you can edit line 159 of config.py in lbuild with this:
`cpath = Path(ConfigNode._rel_path(node.text.replace(":", "_"), configpath))`
I also had a problem with multiprocessing running the examples_compile.py script (another windows related issue, probably deserves another Issue #) which was solved by adding entry point protection. See here: https://github.com/operativeF/modm/tree/fix/win_multiprocessing

salkinium, 2019-02-13:
Can you check if this change in lbuild fixes your issue too? I just wrapped this into an exception.

operativeF, 2019-02-13:
Yup, that fixes it. Thanks

closed 2019-02-13 by salkinium