modm-io/modm #79: Nucleo L3 programming fails

strongly-typed, 2018-09-07:
Working example modm at 4ae989ee0
```
scons program verbose=1
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
openocd -s "/XXXXXXXXXXXXXXXXXXX/src/modm/openocd" -f "board/st_nucleo_f3.cfg" -c "init" -c "reset halt" -c "flash write_image erase build/src.elf" -c "reset halt" -c "mww 0xE000EDF0 0xA05F0000" -c "shutdown"
Open On-Chip Debugger 0.10.0+dev-00449-g4999c9d9 (2018-07-06-10:23)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v18 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.262426
Info : stm32f3x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
adapter speed: 950 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0800180c msp: 0x20000be0
auto erase enabled
Info : device id = 0x10016438
Info : flash size = 64kbytes
Info : Padding image section 1 at 0x080001e8 with 8 bytes
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000003a msp: 0x20000be0
Warn : no flash bank found for address 0x10000000
wrote 14336 bytes from file build/src.elf in 0.807490s (17.338 KiB/s)
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
adapter speed: 950 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0800180c msp: 0x20000be0
shutdown command invoked
scons: done building targets.
```

Non-working example, after updating modm to ea7a3008a
```
scons program verbose=1
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
openocd -f "XXXXXXXXXXXXXXX/src/modm/openocd.cfg" -c "modm_program build/release/src.elf"
Open On-Chip Debugger 0.10.0+dev-00449-g4999c9d9 (2018-07-06-10:23)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
program_debug
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v29 API v2 SWIM v18 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.264000
Info : stm32f3x.cpu: hardware has 6 breakpoints, 4 watchpoints
[... 158 lines cut ...]
in procedure 'program' called at file "/XXXXXXXXXXXXXXXXXXXXXXXXXX/src/modm/openocd.cfg", line 4
in procedure 'program_error' called at file "embedded:startup.tcl", line 522
at file "embedded:startup.tcl", line 477
scons: *** [openocd_run] Error 1
scons: building terminated because of errors.
```

```
cat modm/openocd.cfg
add_script_search_dir modm/openocd
source [find board/st_nucleo_f3.cfg]
proc modm_program { SOURCE } {
	program $SOURCE verify
	reset halt
	mww 0xE000EDF0 0xA05F0000
	shutdown
}

proc program_release {} {
	modm_program build/release/src.elf
}
proc program_debug {} {
	modm_program build/debug/src.elf
}
```

rleh, 2018-09-07:
@chris-durand got a similar error on a STM32F7 board some days ago.
I think it was a OpenOCD related bug. Programming actually works, only verification fails.

strongly-typed, 2018-09-07:
Yes, can confirm that even with the old command (`flash write_image`) the `verify_image` does not work. Now running OpenOCD `Open On-Chip Debugger 0.10.0+dev-00527-g724fddcb (2018-09-07-12:21)`

rleh, 2018-09-07:
Does the flash verification work with `OpenOCD 0.10.0+dev-00527-g724fddcb (2018-09-07-12:21)`?

strongly-typed, 2018-09-07:
No, it doesn't.

rleh, 2018-09-07:
Removing the verify argument from the OpenOCD command would fix the issue:
```diff

proc modm_program { SOURCE } {
-	program $SOURCE verify
+	program $SOURCE
	reset halt
```

But we better investigate why OpenOCD is trying to verify (and program?) SRAM at `0x10000000`.

salkinium, 2018-09-15:
Hm, I thought verifying would make programming more safe, but if this is the broken status of OpenOCD, then maybe we should remove it?

salkinium, 2018-10-05:
Ping?

salkinium, 2018-10-07:
I'm closing this, please open a PR if this OpenOCD bug is too annoying.

salkinium, 2018-10-07:
HA! This isn't a bug in OpenOCD 🤦‍♂️.

This happens when placing the vector table into RAM (which is done by default on F3/F7) and the linkerscript didn't define the `.vector_ram` sections as `(NOLOAD)`, so OpenOCD tried to verify it, which is of course difficult, since the memory is only copied during startup. PR is in works.

PR #97 (2018-10-07): [core] Make default the fastest vector table location
There's no need for the "fastest" option, since this can simply be made the default.
This now also allows other modules to know the real vector table location.

Also fixes #79 by marking the `.vector_ram` section as `(NOLOAD)`.

cc @strongly-typed @rleh @chris-durand
closed 2018-10-07 by salkinium