modm-io/modm #80: Driver for I2C port multiplexer TCA9548A

strongly-typed, 2018-09-07:
@salkinium Yes, a nightmare for our beloved I2C bus.

There is a switch [TCA9548A](http://www.ti.com/lit/ds/symlink/tca9548a.pdf) suitable and made for multiplexing I2C.

I am implementing a driver for this, and I would like to discuss what the best approach is to integrate this into the I2C infrastructure in modm?
The best would be totally transparent to the user, just accessing up to eight independent I2C bus masters, e.g.
```
using I2cMasterMultiplexer = modm::tca9548a<modm::platform::I2cMaster2>;
modm::Vl53l0< I2CMasterMultiplexer[0] > distance0(data0);
modm::Vl53l0< I2CMasterMultiplexer[1] > distance1(data1);
...
```

Any suggestions?

salkinium, 2018-09-07:
[image]
Our I2C implementation needs to 🔥.

You'd have to implement the I2cMaster Transaction state machine on top of a I2c Transaction state machine and then provide 8 access ports to it.
You'd then wrap this in a shim layer like GpioExpander, so you can bind the Tca9548a object to a static type.
```
modm::Tca9548a<modm::platform::I2cMaster2> multiplexor;
using Port0 = modm::tca9548a::Multiplexor<multiplexor, 0>;
```

You'll most likely need to consider atomics at this point, since your Tca9548a implementation will have to be synced with its interrupts. Don't worry about AVRs for this, just use std::atomics and enable the driver only for Cortex-M.

strongly-typed, 2018-09-07:
Sounds like fun, working on it!

strongly-typed, 2018-09-08:
> You'll most likely need to consider atomics at this point, since your Tca9548a implementation will have to be synced with its interrupts

Prefixing every I2C transaction with a transaction addressed to the PCA to select the channel seems to work. The underlying I2C master is queuing I2C transactions (and not reordering!) and sending them in order.

Todos:
 - [ ] Make PCA9548 a template parameter to `i2c_multiplexer.hpp`
 - [ ] Discuss the need for `RF_CALL_BLOCKING`
 - [ ] Only return true in `I2cMultiplexerChannel` when both transactions were queued successfully.

PR #81 (2018-12-03): PCA9548A driver and I2C multiplexed master.
Fixes #80. Open for discussion.
closed 2018-12-03 by salkinium