modm-io/modm #83: Executing lbuild build after 8708f3dc

se-bi, 2018-09-16:
I am facing some trouble with the latest changes introduced by 8708f3dc,
when executing `lbuild build`.

That occurs with various modm examples.

Is it a Python3.5/3.6 issue?

```
  File "/home/user/.local/lib/python3.5/site-packages/lbuild/parser.py", line 110, in parse_repository
    repo = repository.Repository.parse_repository(repofilename)
  File "/home/user/.local/lib/python3.5/site-packages/lbuild/repository.py", line 64, in parse_repository
    repo._functions['init'](RepositoryInitFacade(repo))
  File "/home/user/rca/software/modm/repo.lb", line 164, in init
    name = re.search("<module>(modm:board:.*?)</module>", config.read_text())[1].replace("modm:", "")
TypeError: '_sre.SRE_Match' object is not subscriptable
```

salkinium, 2018-09-16:
This searches through all the board `board.xml` files to extract the name of the board module. Do you have any custom boards there? Can you add a `print(config)` before this line?

salkinium, 2018-09-16:
Ah, no, this is a Python 3.5 issue, where I should use `.group(1)` instead of `[1]`. Ooops.

salkinium, 2018-09-16:
If this fixes it, do you want to open a quick PR?