modm-io/modm #95: Using I2C on Nucleo F303K8

Beetix, 2018-10-05:
Hi,

I am trying to use I2C on the Nucleo F303K8. I would like to run the ssd1306 example from the stm32f4_discovery display examples. 

I am still using I2C1 but instead of GPIOB9 and GPIOB8 I am using GPIOB7 and GPIOB6. I have copied the code (main.cpp and modules from project.xml) over to the blink example of the Nucleo F303K8. The problem is that it does not recognizes I2cMaster1 when compiling. I have tried adding ":platform:i2c:1" as a parameter of `module.depends` function call inside the Nucleo's module.lb but this leads to a dependency resolution error.

Could you help me on this matter?

Thank you

salkinium, 2018-10-05:
There are three implemententations of hardware I2C on the STM32 series and we didn't yet implement the one on F3, that's why the `modm:platform:i2c` module is not available on this target.
Look at the `lbuild discover` output so see which modules and options are available for your specific target.

The workaround right now is to use the `modm:platform:i2c.bitbang` module instead:
```cpp
using Scl = GpioB6;
using Sda = GpioB7;
using Master  = BitBangI2cMaster<Scl, Sda>;
// ...
Master::connect<Scl::BitBang, Sda::BitBang>();
```

salkinium, 2018-10-05:
Just checked, the ["extended" I2C implementation is missing for all F0, F3 and F7 devices](https://github.com/modm-io/modm-devices/blob/develop/tools/generator/dfg/stm32/stm_peripherals.py#L298-L325).

Beetix, 2018-10-05:
Ok, thanks for your fast reply. Is the work on the "extended" implementation of the hardware I2C ongoing work?

salkinium, 2018-10-05:
> Is the work on the "extended" implementation of the hardware I2C ongoing work?

I've developed a bit of a distaste for hardware I2C and it's incredibly stateful (and buggy) implementations on AVR and STM32. The BitBang I2C was so much simpler to write. So probably not in the near future.

Beetix, 2018-10-05:
Ok understood. I confirm that the SSD1306 example is working on the Nucleo F303K8 using BitBang I2C.

closed 2018-10-05 by Beetix