modm-io/modm #98: Python FutureWarning

strongly-typed, 2018-10-08:
```
root@b388c7a65654:/build/src# lbuild build
/usr/local/lib/python3.6/dist-packages/lbuild/config.py:138: FutureWarning: The behavior of this method will change in future versions. Use specific 'len(elem)' or 'elem is not None' test instead.
  if cache_node:
modm requires at least lbuild v1.4.0, please upgrade!
    pip install -U lbuild
```

rleh, 2018-10-08:
I can confirm this warning on *Fedora 28*.

The first is a lbuild warning, the second message is generated by modm.

salkinium, 2018-10-08:
Yes, the `repo.lb` checks for the right lbuild version, since there are small feature additions from time to time.

salkinium, 2018-10-08:
This is related to 
https://github.com/modm-io/docker-modm-build/issues/2.

Please note that I’m  maintaining the docker container for the CI only and not for developer use. I wouldn’t mind though if you wanted to polish it a little. 😜

strongly-typed, 2018-10-08:
Thanks, I'm using it for the CI of our modm projects as well. Was fine so far. Latest update (three weeks ago ...) of modm broke our CI. So I was investigating today. With the updated docker container this messages should be gone. I will close when this assumption hold true.

strongly-typed, 2018-10-08:
@rleh Update your lbuild and modm. Then I cannot reproduce this message. Docker image was updated without error and my CI is green again. Thanks for you help!

closed 2018-10-08 by strongly-typed