modm-io/modm #101: Add I2C support for whole STM32 family

strongly-typed, 2018-10-14:
There has been some work done in roboterclubaachen/xpcc#350 by @rleh 

The STM32L4 driver (which is an ugly name, blame @strongly-typed), seems to be suitable also for F0, F3 and F7. I discussed with @chris-durand on Friday. I have F3 hardware here to test this.

There is the distinction between
* stm32
* stm32-extended
* stm32l4
in  [modm-devices](https://github.com/modm-io/modm-devices/blob/develop/tools/generator/dfg/stm32/stm_peripherals.py#L298).

How integrate this to modm?

chris-durand, 2018-10-14:
I looked at the reference manuals yesterday. `stm32-extended` and `stm32l4` are in fact the same implementation. The memory maps and features seem to be completely identical except for the `WUPEN` bit (Wakeup from Stop mode enable) in `I2C_CR1` that is missing on some controllers (some F0, F7 ?).

I would suggest to rename the L4 I2C to `stm32-extended` and enable it for all the other controllers. That would need some changes in the L4 I2C driver and device file generator. The name `stm32-extended` is problably ok because it supports more features than the other I2C IP (Fast Mode Plus support).

chris-durand, 2018-10-14:
It's not that easy, thank you ST:
[image]

I hope this is somewhere in the CubeMX data…

rleh, 2018-10-14:
[image]
The `WUPEN` bit is not used by the modm anyway, so we can simply ignore it.

I remember that the digital noise filter `DNF` is not available on all controllers.

chris-durand, 2018-10-14:
> I remember that the digital noise filter DNF is not available on all controllers.

I think this is only true for the other IP. If I didn't overlook anything the `DNF` bits are available in all controllers that have the extended IP.

chris-durand, 2018-10-14:
@rleh Are you currently working on this? It wouldn't be very efficient if we end up with 2 PRs implementing the same thing.

rleh, 2018-10-14:
[image]
[image]
At least the feature tables are similar in the reference manuals...

chris-durand, 2018-10-14:
According to CubeMX every device with the `stm32-extended` IP supports FM+ except I2C instance 2 on devices that match `STM32F030C8Tx|STM32F030R8Tx|STM32F051R8Tx|STM32F051C8Tx|STM32F051R8Ux|STM32F051C8Ux|STM32F0x0_Value_Line`.

All instances of `stm32-extended` support SMBus except I2C2 on F0/L0.

This data is somehow contradictory to the reference manuals. STM32F05xxx have no FM+ but I can configure it in CubeMX for STM32F058x8. The correct rule according to the reference manual is that every instance supports FM+ except for I2C2 on F03x, F05x and F070xB devices.

strongly-typed, 2018-10-14:
Is there a way in lbuild to see if there are multiple instances of the peripheral?

in [module.lb](https://github.com/modm-io/modm/blob/develop/src/modm/platform/i2c/stm32l4/module.lb#L46):
```
properties["aid"] = "2" if self.instance == 4 else "1"
```

but ST sometimes leaves out the postfix `1` if there is only one peripheral.

Or should I just create an alias between `APB1RSTR` and `APB1RSTR1`?

salkinium, 2018-10-14:
> Is there a way in lbuild to see if there are multiple instances of the peripheral?

You mean like this?
```py
has_instances = "instance" in device.get_driver("i2c")
```

This is just a dictionary, so you can always use the `get(key, default)` function:  `device.get_driver("i2c").get("instance", [])`.

salkinium, 2018-10-14:
The above is an issue in the CAN drivers, where there's CAN or  CAN1 and/or CAN2. We follow the reference manual naming here, since we don't want to rename stuff (that gets you into trouble).

chris-durand, 2018-10-14:
> in [module.lb](https://github.com/modm-io/modm/blob/develop/src/modm/platform/i2c/stm32l4/module.lb#L46):
>
> ```
> properties["aid"] = "2" if self.instance == 4 else "1"
> ```
>
> but ST sometimes leaves out [...]

This is probably not the right solution to this problem. This relation only holds for some L4. For instance, STM32F745 has 4 I2Cs but no `APB1RSTR1`.

chris-durand, 2018-10-14:
> Or should I just create an alias between APB1RSTR and APB1RSTR1?

L4 controllers have `APB1RSTR1`/`APB1RSTR2`, all the others `APB1RSTR`. That should be manageable.

strongly-typed, 2018-10-14:
[There](https://github.com/modm-io/modm/blob/develop/src/modm/platform/i2c/stm32l4/i2c_master.hpp.in#L82) is a little FIXME we/I need to fix. Blame me :-O

The timing blob was taken from the CubeMX. Does anybody know how to calculate these settings? Is a LUT with a bunch of common frequencies an intermediate solution?

salkinium, 2018-11-06:
Closed in #118.

closed 2018-11-06 by salkinium