modm-io/modm #111: No module named 'modm.parser'

rleh, 2018-10-24:
```
lbuild build
No module named 'modm.parser' 

You might be missing the git submodules in `ext/`.
Try:
    cd modm
    git submodule update --init
then build again.
```

rleh, 2018-10-24:
@salkinium

rleh, 2018-10-24:
```
$ cd modm/
$ git s
On branch develop
Your branch is up to date with 'origin/develop'.

nothing to commit, working tree clean
```

salkinium, 2018-10-24:
Did you by any chance do `pip3 install modm`? Cos modm-devices is conflicting with that, and it sucks.

rleh, 2018-10-24:
Hmm, `pip3 uninstall modm` fixes this problem

closed 2018-10-24 by rleh

salkinium, 2018-10-24:
This is a legacy bug in modm-devices, because the parser is in `ext/modm-devices/tools/device/modm/` and thus is shadowed by the global modm module.
I'll rename this folder to `modm_devices` and adapt the DFG code.