modm-io/modm #112: support L443CCT6

L928, 2018-10-27:
I would like to have L443CCT6 supprted.
Maybe I can help ... but need some assitance where to start.

salkinium, 2018-10-28:
The STM32L443CCT target is already supported (all L4* are) and [compiles successfully on our CI](
https://6913-77459564-gh.circle-artifacts.com/0/log/stm32l443cct.log).
You can `lbuild discover -n :target` to get the full list of supported devices.

Did you mean you want a board support package for a Nucleo board with that target? If I’m not mistaken, there is [only a Nucleo board with the L433rc](
https://www.st.com/en/evaluation-tools/stm32-mcu-nucleo.html), not the L443cc.

salkinium, 2018-10-31:
I'm closing this since the target should work:

```xml
<option name=":target">stm32l443cct</option>
```

Feel free to reopen if you have further issues.

closed 2018-10-31 by salkinium