modm-io/modm #113: DAC API

se-bi, 2018-10-27:
Maybe someone also missing that and implemented it?
@chris-durand ;-)

W-M-D, 2020-03-10:
Ever make any progress on this? I'm about to start implementing it if not.

rleh, 2020-03-10:
Ping @chris-durand :wink:

W-M-D, 2020-03-16:
Well starting now. I'm going to at least do the stm32.

gueldenstone, 2020-05-09:
Hi, I'm a new modm user and I would like to try it on a project using a DAC I see some Code for the mcp4922 in the repository but no way to use it currently @W-M-D is that what you're working on?

salkinium, 2020-05-09:
This issue is about the STM32 integrated DAC unfortunately.

The MCP4922 was part of xpcc.io but I didn't port it to modm, due to running out of motivation and not having the hardware to test it.
If you want to port it, you can have a look at the DRV832x driver, which uses the non-blocking, multiple-access API: https://github.com/modm-io/modm/blob/develop/src/modm/driver/motor/drv832x_spi_impl.hpp#L125-L176

gueldenstone, 2020-05-09:
Oh, I'm sorry anyway thank you for the fast answer! I will look into it and share it if I come up with something :)

PR #420 (2020-06-19): STM32 DAC Platform Driver
Implement a basic synchronous DAC driver for the STM32. 

Closes #113.
closed 2020-06-19 by salkinium