modm-io/modm #125: Support separate compile flags per module/file

salkinium, 2018-11-12:
Some external code may not adhere to the modm coding standards and may generate warnings which may lead to errors due to our general compile flags. This is especially annoying when the external source code cannot be edited due to loosing upstream compatiblity or violating licencing terms.

It would be useful to compile parts of modm and other repositories with different compile flags.
This needs to be done carefully, since some flags are required for the platform (like `flags.archflags`) and must be used for all files (even in other repositories) and some flags may even contain conflicting definitions for separate files (like `flags.cppdefines`).
It is unclear how to define the semantics for choosing what build flags to use for user code, since currently all files in SCons's scope are build with the *union* of all repos/modules of all compile flags.

This will require a new feature in lbuild, for defining `metadata` per file too, not just per module.

salkinium, 2018-11-12:
cc @rleh @strongly-typed @daniel-k @dergraaf @chris-durand @mhthies

dergraaf, 2018-11-13:
> This will require a new feature in lbuild, for defining metadata per file too, not just per module.

A big objection from my side. If we add separate compile flags per file, we end up where we started with the xpcc build system. This is out of scope for lbuild.

> This is especially annoying when the external source code cannot be edited due to loosing upstream compatiblity or violating licencing terms.

The better solution is then to try to fix the issue upstream or wrap the code into a compatible wrapper.
My suggestion would be to separate problematic libraries as much as possible (different folders) and only have different compile flags at that scope (and try to avoid it as much as possible).

salkinium, 2018-11-13:
>> This will require a new feature in lbuild, for defining metadata per file too, not just per module.

> A big objection from my side. If we add separate compile flags per file, we end up where we started with the xpcc build system. This is out of scope for lbuild.

The API would just extend the file operations and attach the metadata to a buildlog operation:

```py
env.copy(..., metadata={"key":"value"})
env.template(..., metadata={"key":"value"})
env.archive(..., metadata={"key":"value"})
```

lbuild already [stores metadata by module](https://github.com/modm-io/lbuild/blob/develop/lbuild/buildlog.py#L87-L117) and provides accessors for global, repository or module level. This would just extend the concept to file operations as well.

The interpretation of this metadata is up to the build script generator in modm, it already produces one separate `SConscript` per repository using this mechanism and includes that into the global `SConstruct`. Note, that "compile flags" are identified by modm as [keys starting with `flags.` and then following the SCons naming convention](https://github.com/modm-io/modm/blob/70fc500bc818ad921736eb952c3786944dded676/tools/build_script_generator/common.py#L67). (These conventions needs to be documented better, but until now they were an internal API).

> The better solution is then to try to fix the issue upstream or wrap the code into a compatible wrapper.

This is unrealistic if there isn't an upstream, or there are simply too many issues to fix.
For example, ST's "upstream" is their website of zipfiles, they neither have a bug tracker, nor can you open pull requests. Similarly the [Invensene-eMD library](https://github.com/modm-io/Invensense-eMD) is also a zipfile, without any way of upstreaming changes.

> My suggestion would be to separate problematic libraries as much as possible (different folders) and only have different compile flags at that scope (and try to avoid it as much as possible).

Unfortunately a lot of libraries don't namespace their CPP defines properly and you may get conflicting definitions globally, especially for defines that are very generic, like [`REMOVE_LOGGING`](https://github.com/modm-io/Invensense-eMD/blob/fe92775f9b56e219949c571205e1ecc2bb11bf57/eMD/driver/module.lb#L35) or anything with `DEBUG` in the name. 🙄

I opened this issue in modm (not in lbuild), since the addition of the metadata per file in lbuild isn't complicated, but the *interpretation* of this data in the build script generator is pretty undefined. For SCons we already compile repos into their own static library and link them all together, however, still compile everything with the *union* of compile flags. That may create obvious issues with conflicting compile flags.

Looking further at the @roboterclubaachen code and users, there are significant requests for features that a typical package management would provide (since we have a lot of repos). I don't want to turn lbuild+modm into a package manager (or use existing ones, since none of them really support embedded crosscompilation), but we may just be forced to provide parts of that API, particularly if we can just "pipe" this through lbuild using its metadata mechanism.

dergraaf, 2018-11-24:
>  but the interpretation of this data in the build script generator is pretty undefined.

This is my main issue with this. It makes the build process much more complicated. Basically it becomes impossible to build modm without using the build script generators (which was my original idea for lbuild).

Unfortunately I don't have a good solution for the problem at the moment.

I would suggest to use this feature as little as possible, e.g. by only having different compilation flags per top-level folders (`ext` or `src`).

salkinium, 2018-11-26:
> Basically it becomes impossible to build modm without using the build script generators (which was my original idea for lbuild).

That's a pretty generic issue tho, I mean, most projects require certain build flags (for CPP defines at the very least) and they're usually bound to a specific build system (CMake for C++ apparently).
I don't think there is any way around this complexity unfortunately, unless we want to delegate the build process entirely to another system like the Conan package manager (which uses CMake).

> Unfortunately I don't have a good solution for the problem at the moment.

I would add another template to the `modm:build`, that generates a markdown file that describes exactly what files needs to be compiled with what flags, what include paths etc.
It'll be like a text-only version of the SCons generator, so it serves the purpose of documenting the modm build system for 1) new build script generators, and 2) manually including modm into your own custom setup (a one-time thing).

salkinium, 2018-12-06:
Now it’s getting interesting: Qt Creator 4.8 was just released with support for \*drumroll\* Compilation Database: 
http://blog.qt.io/blog/2018/12/06/qt-creator-4-8-0-released/

It seems to be a good option in general for generic IDE support, however, it’s still unclear to me if this works with GCC.

siy, 2018-12-07:
@salkinium I've tried compilation DB with gcc and CLion. It works perfectly in regard to code navigation/editing/refactoring. Not tried to compile anything though (I did just need to dig deep into that project code).

closed 2019-07-23 by salkinium