modm-io/modm #128: No module.lb for stm32-f0 adc

cajt, 2018-11-19:


salkinium, 2018-11-20:
Huh? You‘re right! How did that happen?
I even kept the drivers up-to-date regarding connect and pin channel APIs.
I don‘t think there is a reason, maybe this just slipped through the cracks.

salkinium, 2018-11-20:
Ah, I think this was only added in xpcc recently and I may have forgotten to port it correctly.

salkinium, 2018-11-21:
For reference, [this is the original PR, that adds F0 ADC support](
https://github.com/roboterclubaachen/xpcc/pull/327). @strongly-typed suggested that the driver could be merged into F3, due to it‘s similarities, which probably why I haven‘t ported this.

cajt, 2018-11-21:
Acristoffers actually writes that the f0 was derived from f3, but more or less a rewrite. See same PR discussion.
A diff of the adc.hpp.ins looks like it would cause some more "%%if", but also in new spots. And the enums used by the interface are not directly interchangeable.

salkinium, 2018-11-21:
Anyone interested in porting/merging the F0 ADC driver?
cc @chris-durand @rleh ?

chris-durand, 2018-11-21:
> Anyone interested in porting/merging the F0 ADC driver?
cc @chris-durand @rleh ?

Maybe, I'll have a long train ride on Friday, but also have a lot of other things on my TODO list. I can't promise I'll have time for it.

rleh, 2018-11-21:
> Anyone interested in porting/merging the F0 ADC driver?

Unfortunately I currently have too many things on my TODO list. E.g. DMA...

rleh, 2018-11-21:
@cajt Do you need the F0 ADC driver soon?

It might be less effort to add a `module.lb` file to the existing F0 ADC code than to merge the drivers.

cajt, 2018-11-22:
I actually needed it two days ago.
And here it is, but still needs more testing: https://github.com/cajt/modm/tree/fix-stm32f0-adc 

A question came up: Should "Adc" become "Adc1"?

cajt, 2018-11-22:
According to the STM reference manual it is: 

- ADC1, ADC2 and ADC3 for F4, (page 66 in RM0090) and 
- ADC for F0x1/x2/x8 (page 47 in RM0091)

I expect to test this fix on another board and F0 variant, I would suggest to delay the PR until after that test (in a few days-weeks).

salkinium, 2018-11-22:
Oh, ST defines the registers for the [instance as `ADC1` and uses `ADC` for the common registers](
https://github.com/modm-io/cmsis-header-stm32/blob/master/stm32f0xx/Include/stm32f042x6.h#L687). 🙄 Consistency™ by ST.

salkinium, 2018-11-22:
In the long term, I‘d like to decouple instance naming from their interfaces. For example USART1 can be used as either UART or SPI, so there should be `modm::platform::Usart1::Uart` and `modm::platform::Usart1::SpiMaster` classes and the shared „HAL“ implementation should be in modm::platform::Usart1`. Here it would become `modm::platform::Adc*::Adc` and `modm::platform::Adc*::AdcInterrupt`.

cajt, 2018-11-25:
In the meantime I also got it working on a custom board using a F072. The results look reasonable, but its far from full coverage of everything the ADC can do.

Shall I submit a pull-request?

salkinium, 2018-11-25:
Yes, please, it make it easier for the next developer to continue this work.

PR #132 (2018-11-27): STM32F0 ADC Support
Added support for STM32F0 ADC
Only basic testing on F042 and F072.
Add example for STM32F0 ADC
Add copyright notice

Closes #128.
closed 2018-11-27 by salkinium