modm-io/modm #140: Rename systemClock::Frequency to systemClock::sysClk

squarecode, 2019-01-07:
The term Frequency is really misleading in many cases. 

[image]

Using the term used by ST would be pretty cool for me as a beginner.

salkinium, 2019-01-08:
The entire SystemClock needs to be refactored to match ST‘s documentation. For example, `SystemClock::TimerN` should be called `SystemClock::TimN`.

patrick--, 2019-02-10:
Would you want all the refactoring of the names to be done in a single PR? If so what would that complete list look like? Otherwise I can get started with at least `SystemClock::TimN` and `SystemClock::sysClk`

salkinium, 2019-02-10:
This sounds easy, but it’s unfortunately a very complicated issue. The background is that we don’t yet have the [clock tree information in modm-devices](
https://blog.salkinium.com/modm-devices/#clock-tree), so we had to manually create all the [SystemClock structures for the board support packages](
https://github.com/modm-io/modm/blob/develop/src/modm/board/disco_f303vc/board.hpp#L30-L109). The usability of this obviously sucks, since you have to look up what peripherals are available manually in the datasheets and what frequencies are acceptable for what part of the tree.

Back in [2015 I prototyped a C++ compile time API](https://github.com/salkinium/save-the-clocktrees) for a better system clock abstraction, but it turned out to be quite difficult to map a graph into a useful template structure with all the edge cases still working. I abandoned this concept.

The way I would like to solve this issue is to parse the clock tree information in CubeMX via modm-devices and at least generate a base class that contains the correct mapping of clock domain -> peripheral (ie. I2C1 is connected to APB1 or APB2?) and maybe abstracts some domain prescalers away.
The BSP package would then inherit from this description and specialize it, in order to cut down on the boilerplate code.

Something like this maybe:
```cpp
template< uint32_t SysClk, uint8_t AhbPrescaler = 1, uint8_t Abp1Prescaler = 1 >
struct SystemClockBase
{
    static constexpr uint32_t SysClk = SysClk;
    static constexpr uint32_t Ahb = SysClk / AhbPrescaler;
    static constexpr uint32_t Apb1 = Ahb / Abp1Prescaler;
    // etc, add all device specific peripherals here:
    static constexpr uint32_t Tim1 = Apb1;

    bool
    initPrescalers()
    {
        // set AHB, APBx prescalers
    }
    
    bool
    initSysClk()
    {
        // init Flash latency etc
    }

    bool
    InitClock()
    {
        // set modm::clock::fcpu etc values
    }
};
// in BSP or user code
class SystemClock : public SystemClockBase<16_MHz>
{
    // constanst are all inherited
    bool
    enable()
    {
        initPrescalers();
        // setup PLL
        initSysClk();
        initClock();
    }
};
```

closed 2021-09-29 by salkinium