modm-io/modm #143: Change: "nrf24hal" to "nrf24phy" in nrf24-scanner example

FadiBunni, 2019-01-10:
The nrf24-scanner example, has "nrf24hal" as a function to be called, but it does not exists. It should be changed to "nrf24phy"

salkinium, 2019-01-10:
> I never done it before, but would like to know how.

Possibly the easiest way is to do it online:
https://help.github.com/articles/editing-files-in-another-user-s-repository/

Here is the [file in edit mode](https://github.com/modm-io/modm/edit/develop/examples/stm32f4_discovery/radio/nrf24-scanner/main.cpp) to get you started.

salkinium, 2019-01-10:
For more complicated changes, you should commit your changes to a separate branch on your fork and manually open the PR:
https://help.github.com/articles/creating-a-pull-request/

FadiBunni, 2019-01-10:
Alright, I made the simple change for now, but will use the guide for complicated changes, when there is more stuff. 

ps: I sent you a mail :)

salkinium, 2019-01-13:
Do you want to open a PR with your changes? 
https://github.com/FadiBunni/modm/commit/ec0c030e91817edf9da38c3d96b340c7abcdd051

salkinium, 2019-01-18:
Fixed in #149

closed 2019-01-18 by salkinium