modm-io/modm #148: TODOs and Ideas

salkinium, 2019-01-18:
Some rough ideas about what needs to be done without having to open an individual issue for each one of them.

## Library

- [ ] Run unittests on AVR and STM32 targets
- [ ] Fix AVR unittests
- [x] Move bitmap generation into it's own module.
- [ ] Move XPCC protocol out of modm into its own `xpcc-communication` repository.
- [ ] Update FreeRTOS to v10 with MIT license, see `feature/freertos` branch
- [ ] Update FatFS to latest version
- [ ] Use font generation scons tool for generating `modm/ui/display/font/*`
- [ ] Check why `modm::BlockAllocator` and `modm::allocator::Block` exist
- [ ] choose default allocator by available RAM size:
  - `block` for < 8kB total RAM 
  - `newlib` for 1 continous region (+ bus fault recovery)
  - `tlsf` for multiple discontinous regions
- [ ] Let user choose which memory regions to use as heap
- [ ] Choose default `minimum_pool_size` by available RAM size for TLSF allocator
- [ ] assign linkerscript sections via memory attributes
- [ ] call same function on a set of pins (ie. access-optimized `setAlternateFunction()`)
- [ ] Use CCACHE with SCons tools? Initial experiments didn't trigger CCACHE re-use!
- [ ] Implement ITM logging for boards without serial
- [ ] Windows support needs work

## SCons

- [ ] Add programming capabilities
  - [x] openocd
  - [ ] BMP
  - [ ] DFU
  - [x] avrdude
- [ ] fix avr-size
- [x] Add debug/release profile
- [ ] Font generator?
- [x] Upgrade to SCons 3 (done for all modm scons tools)
- [x] XPCC communication generator
- [x] Improve size tool output
- [x] Unittests
- [x] hosted compilation

## CMake

- [ ] Multi-repository builds
- [ ] Individual compile flags per file

ASMfreaK, 2019-08-12:
I saw some time ago discussion on code formatting.
I saw [this tool](https://zed0.co.uk/clang-format-configurator/) on C++ Weekly, enabling one to choose and save clang-format configuration.
I'm not sure, if we'll be able to make great use out of it, since a lot of modm code is Jinja templates, unparseable by clang.

parzival2, 2020-03-12:
Any plans to add USB support?

salkinium, 2020-03-12:
> Any plans to add USB support?

Yes, preferrably via third-party project integration though, since doing USB properly seems fairly complicated. I've bookmarked these projects to evaluate:

- https://github.com/hathach/tinyusb
- https://github.com/dmitrystu/libusb_stm32
- https://github.com/IntergatedCircuits/USBDevice
- https://github.com/kevinmehall/usb

parzival2, 2020-03-12:
@salkinium Great. If I get around to gain a bit of knowledge into the project, may be I can create a pull-request using any of the above mentioned sources. I just discovered modm yesterday and its great.

parzival2, 2020-03-12:
I saw that ROS is already available but having nanopb(https://github.com/nanopb/nanopb) will also be really great. It will be useful in writing binary files to SD-Card. The idea that I had in mind was to collect Accelerometer, Gyroscope data and play them back to simulate sensors like ROS bags. What dou you guys think?

salkinium, 2020-03-12:
> What do you guys think?

You don't *have* to integrate nanopb into modm directly (in fact I would *prefer* to keep the modm core repo not expanding in size too much). You could make it a separate lbuild repository (like https://github.com/modm-io/Invensense-eMD) or modify the *generated* SCons/CMake build system to also build nanopb. The goal is to not have to "wrap" every library with modm, as modm probably adds very little to those libraries.

> It will be useful in writing binary files to SD-Card

Note that our version of FatFS is very old, it should be upgraded. In addition there isn't that much *explicit* support for SD-Cards or filesystems in general in modm. (sorry)

salkinium, 2020-03-12:
Oh, nanopb comes with both CMake and SCons tools, so maybe it is warranted to create a proper lbuild repository and integrate not only the source but also the protocol builder into the modm build system. Probably not trivial though.

W-M-D, 2020-03-18:
Implementing some SD support soon!

rleh, 2020-03-18:
> Implementing some SD support soon!

Once I added a block device interface (and implementations for SPI flash, heap storage and mirroring (like RAID1) to modm: https://docs.modm.io/develop/api/stm32g474vet7/classmodm_1_1BlockDevice.html
The idea was to use this interface also for SD cards, on-chip flash and other storage devices.

Then file systems like FAT (or ext3) could be implemented via this interface and used for all storage.

Unfortunately this interface has not been used for the recently added Eeprom driver in #321.

mikewolfram, 2020-03-18:
I wasn't aware of the BlockDevice, so I did the CAT24AAxx driver by simply modifying the existing EEPROM driver. I'll look into it and change it.

mikewolfram, 2020-03-18:
Talking about upcoming implementations, I'm interested in DMA and UI. I've been looking in lvgl but wasn't able to get it up and running. Their memory management didn't work on my L431. And also it adds a lot of code, my test of having a simple button added >30k (with -Og setting). I'd rather use something else. The current UI in modm has small footprint and would work perfectly to my needs.

rleh, 2020-03-18:
> DMA

You know there currently exists a very basic and hardly used [DMA API](https://docs.modm.io/develop/api/stm32f407zgt7/group__modm__platform__dma.html)?

@chris-durand and me started to work on DMA two years ago. On my [`feature/dmahal` branch](https://github.com/rleh/modm/commits/feature/dmahal) are a lot of *WIP*-commits. I'll try to rebase the branch soon.

DMA can be used for many different use cases. What is your use case?
Maybe we could discuss the use-cases and thus the interface of the DMA API(s) in a separate issue?

salkinium, 2020-03-18:
>  I'm interested in DMA

[image]

> Maybe we could discuss the use-cases and thus the interface of the DMA API(s) in a separate issue?

Yes please. I've added a [bunch of DMA data already to the device files](https://github.com/modm-io/modm-devices/pull/1) for this purpose!

rleh, 2020-03-18:
For DMA: see #358

jasa, 2020-05-26:
Better sanity checks. After some quick testing i encountered type mismatches across units. Too many warnings is disabled after my taste, which can lead to undefined behaviour. Tools as clang-tidy and cppcheck can be used on the generated code.

Example of code smells:

_platform/fault/fault_storage.cpp:36:17: note: type 'const uint32_t' should match type 'const uint8_t'
   36 | const uint32_t *fault_storage_heap_start{nullptr};_

_platform/core/heap_table.cpp:22:27: warning: type 'struct table_pool_t' violates the C++ One Definition Rule [-Wodr]
   22 | } __attribute__((packed)) table_pool_t;_

_platform/fault/fault_storage.cpp:18:17: note: a field of same name but different type is defined in another translation unit
   18 |  uint32_t const traits;_

salkinium, 2020-05-26:
I completely agree! This only works if it is automated in the CI though. Would you be interested in setting this up (or adding more warnings)?

closed 2021-09-29 by salkinium