modm-io/modm #157: No timers and PWM functions for AVR

ASMfreaK, 2019-02-07:
References #23.

What are the steps towards enabling support for AVR timers?
Is it manageable to add support for timers to a small handful of devices (like mega328p, mega32u4, mega2560)?

salkinium, 2019-02-07:
For any AVR timer code you would implement a new lbuild module in `modm/src/modm/platform/timer/avr/` similar (but less complicated) to the [existing stm32 timer implementation](
https://github.com/modm-io/modm/blob/develop/src/modm/platform/timer/stm32/module.lb) including Timer instances as submodules (`modm:platform:timer:1` etc).
You will want to check the [modm-devices AVR data](
https://github.com/modm-io/modm-devices/blob/develop/devices/avr/atmega-48_88_168_328-a_n_p_pa.xml#L33-L41) for device specific instances and type of peripheral.

You would limit the module to AVR platforms only, or even further to only the few devices that you want to support, by returning True or False in the module’s prepare step:
```py
def prepare(module, options):
    device = options[":target"]
    if not device.has_driver("tc:tc*"):
        return False
    if device.identifier.string not in [“atmega328p”, “atmega32u4”, “atmega2560”]:
        return False
    return True
```
Make sure to provide at least one example for these devices using this timer, otherwise your code may never be compiled on the CI.

As far as the actual C++ API is concerned, modm doesn’t have an interface for Timers, since we’ve always used them for motor control and that application has always been very specific to the hardware. So for the STM32 timers we’ve created a STM32-specific abstraction, which may not be very useful for AVRs. So don’t feel like you need to provide the same interface, if you’ve found something better. For simple use-cases like PWM an architecture interface would be nice to have though.

Just open a PR and we and the CI can guide you along.

ASMfreaK, 2019-02-12:
I'm working on this, soon to release first mwp pr.

closed 2021-09-29 by salkinium