modm-io/modm #158: External projects fail to find modm repository (and docs are lacking)

ASMfreaK, 2019-02-09:
I have a newbie question, which I could not resolve through docs.

Consider the following directory structure:
```bash 
$ tree -L 2
.
├── modm
│   ├── AUTHORS
│   ├── ....
│   └── tools
├── modm_example
│   ├── main.cpp
│   └── project.xml
└── modm_stm_example
    ├── blinky.cpp
    └── project.xml
```

`modm_example/project.xml`:
```xml
<library>
  <extends>modm:board:arduino-uno</extends>
  <options>
     <option name="modm:build:build.path">build</option>
  </options>
  <modules>
    <module>modm:platform:adc</module>
    <module>modm:build:scons</module>
  </modules>
</library>
```
I have my project `modm_example` not in modm repo. How can I build it?

I tried two methods:
1. W/o options
```bash
$ cd modm_example
$ lbuild build
[ERROR] lbuild.parser: 
ConfigNode(filename='project.xml')

ERROR: No repositories loaded!
```
2. Using `-r` option:
```bash
cd modm_example
lbuild -r ../modm/repo.lb build
ERROR: Repository name 'modm' is ambiguous. Name must be unique.
```

For now I'm using the second method. here is the diff:
```diff
diff --git a/src/modm/board/arduino_uno/board.xml b/src/modm/board/arduino_uno/board.xml
index 0272933af..ec3c4a03e 100644
--- a/src/modm/board/arduino_uno/board.xml
+++ b/src/modm/board/arduino_uno/board.xml
@@ -1,13 +1,9 @@
 <library>
-  <repositories>
-    <repository>
-      <path>../../../../repo.lb</path>
-    </repository>
-  </repositories>
-
   <options>
     <option name="modm:target">atmega328p</option>
     <option name="modm:platform:clock:f_cpu">16000000</option>
+    <option name="modm:build:avrdude.programmer">arduino</option>
+    <option name="modm:build:avrdude.baudrate">115200</option>
   </options>
   <modules>
     <module>modm:board:arduino-uno</module>
```

I had the same error with Blue Pill (`<extends>modm:board:blue-pill</extends>`)

salkinium, 2019-02-09:
You have to provide a path to the repos either in your own `modm_example/project.xml` using the `<repository>` elements like in the board config file, or via the command line like you tried.

It seems there is a bug in lbuild which shouldn’t try to parse the same `repo.lb` file twice, regardless of how many times it is declared. For now, I recommend you add the `<repository>` elements to your `project.xml`.
(The examples don’t have this issue, since there is a `modm/examples/lbuild.xml` file, which lbuild automatically finds and it contains the repo path, I’ll update the docs for this).

salkinium, 2019-02-09:
I've fixed the duplicate parsing of repo files in lbuild v1.6.4 just now/

ASMfreaK, 2019-02-09:
I can verify the fix. I now use in parent folder this `./lbuild.xml`:
```xml
<library>
  <repositories>
  	<repository><path>modm/repo.lb</path></repository>
  </repositories>
</library>
```
Works great! It is fixed for me. Should I close the issue?

salkinium, 2019-02-09:
Excellent! That’s exactly how `lbuild.xml` is supposed to be used.

Leave the issue open for now, it reminds me about what I need to clarify.
We’re still working on better docs at https://doc.modm.io, but it’s a bit complicated with all these different devices and options.

salkinium, 2019-02-21:
I've added a tip to the getting started guide regarding the repository paths:
https://modm.io/guide/getting-started/#custom-configuration

closed 2019-02-21 by salkinium