modm-io/modm #170: Support for stm32duino bootloader or linking to different flash address

dergraaf, 2019-02-18:
In order to use the [stm32duino bootloader](https://github.com/rogerclarkmelbourne/STM32duino-bootloader) on a blue-pill board it is required to link the user code to `0x08002000` instead of `0x0800000` (default flash location). Using the currently available lbuild options this seems not to be possible without using a custom linkerscript (which I would like to avoid).

Is this a broad enough use case to merit an additional option to add an offset or a custom flash location for linking?

salkinium, 2019-02-18:
> Is this a broad enough use case to merit an additional option to add an offset or a custom flash location for linking?

Yes, I think this can be a useful special case to have an option for!
It's also easier to document this option than to document how to change the generated linkerscripts.

salkinium, 2019-03-06:
Fixed in #179.

closed 2019-03-06 by salkinium