modm-io/modm #189: Support for ATSAM arm processors

TheSlat, 2019-04-04:
The Atmel (Microchip) ATSAM line of ARM microprocessors are very nice microcontrollers. The SAMD21 is used on Arduino boards. Adafruit has SAMD21 and SAMD51 based feather boards. Peripherals are conceptually similar to the nicer AVR peripherals (SERCOM, timers, etc) and there is a lot of code/pinout compatibility between models.

I actually use the SAML21 processor on a few projects, which is a very low power M0+. I build very low power data-loggers for physiological experiments, and it seems like my use case has a large overlap with modm.

Perhaps I can help add support for these processors?

salkinium, 2019-04-04:
[modm relies heavily on the data in modm-devices to generate the HAL](https://modm.io/how-modm-works/). In order to properly support the Atmel SAM platform, I would first need a data source similar to CubeMX with machine-readable data. (See [modm-devices](https://blog.salkinium.com/modm-devices/)). Otherwise you'd go crazy trying to manually add all this data to a similar degree of completeness and correctness as with the existing AVR and STM32 platforms and it wouldn't really be maintainable for us either.

Do you know of some configuration GUI or similar data source for Atmel SAMs?

salkinium, 2019-04-04:
Bullseye: http://packs.download.atmel.com has SAM* data as well. We already use this for AVR, a quick glance over the `.atdf` files shows interrupt table, pins and pin alternate functions. It also includes the cmsis header files, which we would add as a separate repository (`cmsis-header-sam`?) analog to the [cmsis-header-stm32 repo](https://github.com/modm-io/cmsis-header-stm32).

TheSlat, 2019-04-04:
yes the packs link you found contains the `.atdf` files as well as the standard ARM `.svd` files.
There is also [https://start.atmel.com/](https://start.atmel.com/) which is a web gui that generates projects as well as a machine readable description of that project. I don't know the details but I know Adafruit does some automated generation and download using Atmel Start to maintain their copy of ASF4 ( [https://github.com/adafruit/asf4/tree/master/tools](https://github.com/adafruit/asf4/tree/master/tools) ).

salkinium, 2019-04-04:
I think the `.atdf` files contain enough information to cover all the modm features. Do you want to try to add them into modm-devices?

TheSlat, 2019-04-04:
I'll give it a shot. Looks like I need to do two things:
  * create a CMSIS repo with a script(s) to download the device packs, suck out the headers, and make sure they are formatted properly.
  * fork modm-devices and add the new processors by parsing the `.atdf` files through the existing tools.

I'll probably start with the D21 since the featherD21 is very easy to get and cheap compared to all my custom L21 hardware. If I hit a wall I will post here.

ASMfreaK, 2019-04-05:
Does this also include  Atmel SAM3X8E ARM Cortex-M3 found in Due?

salkinium, 2019-04-05:
I don‘t see a reason that the modm-devices parser cannot parse ALL pack files of ALL SAM* devices. The formats are pretty much all the same (but the devil is in the detail like with the STM32 Cube Data). What the extend of the HAL implementation will be, depends heavily on the fragmentation of the hardware. But modm is very modular, so we can build that up gradually. I‘d focus first on startup, then GPIO, then UART, then the rest.

salkinium, 2019-04-05:
So, yes, why not? Once we have the modm-devices data, everything becomes *significantly* easier to port.

Note that I recently separated the Cortex-M core from the STM32 core, in anticipation of additional Cortex-M based vendors: #176.

salkinium, 2019-04-09:
Please don't hesitate to ask for help if your stuck, since my code is not particularly Pythonic or well documented.

TheSlat, 2019-04-11:
[Header repo similar to modm-io/cmsis-header-stm](https://github.com/TheSlat/cmsis-header-atsam)
just samd51 and samd21 for now. Since Atmel keeps all the packs on one page I think a future feature could be to discover all available SAM devices from the html, instead of using a hard-coded list.

Also packs contain separate trees for each variant. For example SAMD21 has SAMD21A. SAMD21B, SAMD21C... etc. I did diffs on the folders and there are differences so I just made each variant a top level folder.

[fork of modm-devices](https://github.com/TheSlat/modm-devices)
got script to download and extract the `.atdf` files into the expected location (easy)
the dfg folder is a little more confusing, but I made the class for parsing the part number.

salkinium, 2019-04-20:
Ok, we now have data, lets format that into some code using lbuild.
Note that you can explore the available lbuild modules for your target using
`lbuild -r repo.lb -D :target={identifier} discover --developer`.

Here's a checklist to add a new Cortex-M platform to modm:

- [x] Find data source and format into device files in modm-devices and point `ext/modm-devices` submodule to that.
- [x] Extract CMSIS headers into own repository and add as submodule in `ext/{vendor}/{platform}`.
- [ ] Add new platform devices to `modm:target` option in [repo.lb](https://github.com/modm-io/modm/blob/develop/repo.lb#L71).
- [ ] Create a very simple GPIO example for your test device in `examples/{board}/blink` (like [AVR/blink](https://github.com/modm-io/modm/tree/develop/examples/avr/gpio/blinking)).
      This example will be used to test the startup, delay and GPIO port.
      Try out `lbuild discover --developer` and find what is missing in the tree (`modm:platform:*` mostly).
      `lbuild build` will probably fail.

- [ ] Port startup code

`modm:platform:core` module in `modm/src/modm/platform/core/{platform}`:

The startup code is shared for ARM Cortex-M platforms in the [`modm:platform:cortex-m`](https://github.com/modm-io/modm/tree/develop/src/modm/platform/core/cortex).
But you will have to port the linkerscript, and the `__modm_initialize_platform()` function.
Use [the STM32 code module](https://github.com/modm-io/modm/tree/develop/src/modm/platform/core/stm32) to port your own.

The Cortex-M module provides you [with predefined sections](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/cortex/linker.macros) to assemble the linkerscripts.
You need to *arrange* those in the right order and assign these sections to the right memory.
The most restrictive linkerscript is the one for [just one continuous SRAM section](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/stm32/linkerscript/stm32_ram.ld.in).
Note the use of table sections, which tell the startup script what memory to copy/zero and what SRAM to use for the heap.
The heap memory is attributed using [these traits](https://github.com/modm-io/modm/blob/develop/src/modm/architecture/interface/memory.hpp#L27).

`modm:cmsis:device` module in `ext/{vendor}`:

This module provides the CMSIS header files and CPP defines for the specific device.
Note that [the STM32 module](https://github.com/modm-io/modm/tree/develop/ext/st) also does
additional work for aiding debugging and providing clock control definitions.
These are not necessary initially.

(I'll add more here later)

salkinium, 2019-04-23:
Btw, I ordered a different [SAMD21 breakout board](https://robotdyn.com/samd21-m0-mini.html), so I can test your port and add another board.

strongly-typed, 2019-05-23:
> Btw, I ordered a different [SAMD21 breakout board](https://robotdyn.com/samd21-m0-mini.html), so I can test your port and add another board.

Me too! Great to see some porting effort here! I just received an NXP board ...

BTW, I get a
```
scons: *** [__size] ZeroDivisionError : float division by zero
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/scons/SCons/Action.py", line 1202, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "/Users/usr/Dev/modm/modm-the-slat/examples/feather_m0/blink/modm/ext/dlr/scons-build-tools/site_tools/utils_buildsize.py", line 123, in size_action
    "rom_p": totals["rom"] / float(flash) * 100.0,
ZeroDivisionError: float division by zero
scons: building terminated because of errors.
```

when I do a `lbuild build && scons` in `modm-the-slat/examples/feather_m0/blink` at `feature/microchip-sam` (`254eb07`) and `modm-devices` at `ba51890`

tannewt, 2020-04-23:
I've done a bunch of work (CircuitPython) with the SAMD21 and SAMD51 and can help answer questions if folks have them.

salkinium, 2020-04-25:
Thanks for reminding me about this work, @tannewt! I think all the hard stuff for SAMD support is already done, including modm-devices data, so I will work on merging modm-devices and cleaning up the PR and merging it too in the current state. Then others can continue porting GPIO/UART/etc without having to deal with the foundations too.

PR #194 (2020-07-05): Support for ATSAM arm processors
just a start so we can comment. Still wrapping my head around the python-template stuff.
Fixes #189.

Update 2020-04-27: @salkinium took over this PR, see https://github.com/modm-io/modm/pull/194#issuecomment-619634443
closed 2020-07-05 by salkinium