modm-io/modm #192: lbuild generates bad scons files on WIndows

ASMfreaK, 2019-04-09:
On windows, `lbuild build` generates `modm/Sconscript` with wrong file separators like this:
```python
....
files = [
    env.File("src\modm\architecture\driver\atomic\flag.cpp"),
    env.File("src\modm\architecture\interface\can.cpp"),
    env.File("src\modm\architecture\interface\can_message.cpp"),
    env.File("src\modm\board\board.cpp"),
....
 ```
For now, I cope with this using `sed` and replacing all `\` with `/` after `lbuild build`. 
After that `modm/Sconscript` is usable, but I get the same error when trying to run `scons program`, but worse - as it is passing file name and it gets garbled - instead of `build/release/some_name.elf` I get `buildeleaseomename.elf`. It isn't fixable via same method.
Although, using `openocd -f modm/openocd.cfg -c program_release` works after replacing all `\\` in `openocd.cfg` with `/`

salkinium, 2019-04-09:
The thing is, there are apparently several different ways to do this wrong, depending on Windows version as well as Python environment, calling environment. I don't use Windows, so I don't have an overview:

- https://github.com/modm-io/modm/issues/67
- https://github.com/modm-io/modm/pull/69
- https://github.com/modm-io/modm/pull/123

The way to fix this is to [fix the `windowsify` filter](https://github.com/modm-io/modm/blob/develop/repo.lb#L184) and then use that in the SConstruct (and CMake) generators.

salkinium, 2019-04-09:
What Windows version do you use and what environment do you call lbuild and SCons in?

ASMfreaK, 2019-04-09:
I used Windows 7 on python 3.7 in a virtualenv. I used Git for Windows's MSYS bash when creating and running virtualenv, but the problem persisted after switching to launching the same thing from QtCreator.

salkinium, 2019-04-10:
Do you want to propose a fix? Probably adding ` | modm.windowsify` to the file outputs will fix this.

dergraaf, 2019-04-10:
This shouldn't be something which is fixed by adding the `windosify` filter everywhere. In general that filter should not be necessary at all.

Maybe using `pathlib` everywhere can solve this problem? Then we can use `as_posix()` where the tools explicitly require forward slashes under Windows and the normal path object for everything else.

salkinium, 2019-04-10:
There are some files that are not called by Python: The whole CMake generator, the `modm/gdbinit` file and the `modm/openocd.cfg` file. These files are also called by different programs, GDB and OpenOCD, which I assume use the system calls to access the file system. Hence this global filter.

I find all of the online `\ vs. /` discussions confusing, some claim that Windows does work with `/`, others say it only works as of Windows 10. What about that Linux Subsystem thing? I don't have an overview of what works where, so it's very difficult for me to fix.

chechu2k5, 2019-04-11:
a quick workaround (on win10, python 3.7 virtualenv) that worked for me was to mark the file paths as 'raw' python strings like so (notice the ` r ` in front of the path string)
```python
files = [
    env.File(r"src\modm\architecture\driver\atomic\flag.cpp"),
    env.File(r"src\modm\board\board.cpp"),
    env.File(r"src\modm\io\iodevice.cpp"),
    env.File(r"src\modm\io\iostream.cpp"),
    env.File(r"src\modm\io\iostream_float.cpp"),
    env.File(r"src\modm\io\iostream_printf.cpp"),
    env.File(r"src\modm\math\utils\bit_operation.cpp"),
    env.File(r"src\modm\math\utils\pc\operator.cpp"),
    env.File(r"src\modm\platform\clock\rcc.cpp"),
    env.File(r"src\modm\platform\clock\systick_timer.cpp"),
    env.File(r"src\modm\platform\core\assert.cpp"),
    env.File(r"src\modm\platform\core\clock.cpp"),
    env.File(r"src\modm\platform\core\cxxabi.cpp"),
    env.File(r"src\modm\platform\core\default_handler.sx"),
    env.File(r"src\modm\platform\core\delay.cpp"),
    env.File(r"src\modm\platform\core\heap_newlib.c"),
    env.File(r"src\modm\platform\core\heap_table.c"),
    env.File(r"src\modm\platform\core\startup.c"),
    env.File(r"src\modm\platform\core\startup_platform.c"),
    env.File(r"src\modm\platform\core\vectors.c"),
    env.File(r"src\modm\platform\gpio\enable.cpp"),
    env.File(r"src\modm\utils\dummy.cpp"),
]
library = env.StaticLibrary(target="modm", source=files)
```

salkinium, 2019-04-11:
Yes, that’s required because `\` is the escape character. That’s why the windowsify filter replaces the with `\\`.

ASMfreaK, 2019-04-12:
@salkinium 
Unsurprizingly, changing `tools/build_script_generator/scons/resources/SConscript.in`'s like this:

```diff
diff --git a/tools/build_script_generator/scons/resources/SConscript.in b/tools/build_script_generator/scons/resources/SConscript.in
index 298296667..fc18e0bd3 100644
--- a/tools/build_script_generator/scons/resources/SConscript.in
+++ b/tools/build_script_generator/scons/resources/SConscript.in
@@ -128,7 +128,7 @@ env.AppendUnique(CPPPATH=[

 files = [
 %% for file, flags in sources if not flags | length
-    env.File("{{ file | relocate }}"),
+    env.File("{{ file | relocate | modm.windowsify }}"),
 %% endfor
 ]
 %% for file, flags in sources
```

Fixes the compilation problem.

salkinium, 2019-04-15:
Could you also check the CMake generator? I’m pretty sure those paths are not gonna work correctly.

Sh4rK, 2019-05-04:
I've run into this too.

I was looking into the windowsify filter, but I realized that in this case at least, the problem is not that it's not a windows path, the problem lies with the fact that the windows path separator happens to be the same as the escape character in Python string literals (the current code would be equally problematic with unix paths containing backslashes, I think).

What is actually needed is to convert a string to a python-string-literal format before writing it into the template. So basically instead of `env.File("{{ file | relocate }}")` something like `env.File({{ file | relocate | strlit }})` could be used where `strlit` is a filter that does `repr(str(thing))`.

This is basically the same problem as escaping HTML, just doing it with string literals here.

Does this make sense, and seem like a good idea to anyone else?

salkinium, 2019-05-04:
I think what @chechu2k5 mentioned with the raw strings is correct:
`modm.windowsify` should convert the `/` to **one** `\` so that we can format paths in `openocd.cfg` and `gdbinit` correctly, but then for SConstript we use raw strings, so that we don't have to escape the backlashes with `\\`. Then the `modm.windowsify` filter doesn't have to be context aware.

So just this in `repo.lb`:
```py
    if "Windows" in platform.platform():
        repo.add_filter("modm.windowsify", lambda p: p.replace("/", "\\"))
    else:
        repo.add_filter("modm.windowsify", lambda p: p)
```

And then this in SConscript.in everywhere:
```diff
 files = [
 %% for file, flags in sources if not flags | length
-    env.File("{{ file | relocate }}"),
+    env.File(r"{{ file | relocate | modm.windowsify }}"),
 %% endfor
 ]
```

But there are a few paths in SConscript.in that don't seem to have this issue and they are all wrapped in `abspath()`, like at the very top:
```py
env.Append(toolpath=[abspath("scons/site_tools"), abspath("ext/dlr/scons-build-tools/site_tools")])
```
Why doesn't that fail?
Maybe we can just wrap the strings in abspath?

salkinium, 2019-05-04:
Does the gdbinit and openocd.cfg even care about backlashes? I would assume that they probably operate internally on forward slashes since they're OSS.

I mean, the current `openocd.cfg` file does *and* doesn't use the windowsify filters at the same time:
```
%% for path in collector_values["path.openocd"] | sort
add_script_search_dir {{ path }} # <<< no windowsify filter
%% endfor
%% for file in collector_values["openocd.source"] | sort
source [find {{ file }}] # <<< no windowsify filter either
%% endfor

%% for profile, path in elf_files.items()
proc program_{{ profile }} {} {
	modm_program {{ path | modm.windowsify }}
}
%% endfor
```

Sh4rK, 2019-05-05:
> I think what @chechu2k5 mentioned with the raw strings is correct:
> `modm.windowsify` should convert the `/` to **one** `\` so that we can format paths in `openocd.cfg` and `gdbinit` correctly, but then for SConstript we use raw strings, so that we don't have to escape the backlashes with `\\`. Then the `modm.windowsify` filter doesn't have to be context aware.
> 
> So just this in `repo.lb`:
> 
> ```python
>     if "Windows" in platform.platform():
>         repo.add_filter("modm.windowsify", lambda p: p.replace("/", "\\"))
>     else:
>         repo.add_filter("modm.windowsify", lambda p: p)
> ```
> 
> And then this in SConscript.in everywhere:
> 
> ```diff
>  files = [
>  %% for file, flags in sources if not flags | length
> -    env.File("{{ file | relocate }}"),
> +    env.File(r"{{ file | relocate | modm.windowsify }}"),
>  %% endfor
>  ]
> ```

Seems reasonable. I went with the `repr` way because theoretically a file path can contain arbitrary byte values except a few special cases (on linux for example path components can contain anything other than `/` and the zero byte), in which case not every path is representable as a raw string, but it is as a normal string with escapes.

> But there are a few paths in SConscript.in that don't seem to have this issue and they are all wrapped in `abspath()`, like at the very top:
> 
> ```python
> env.Append(toolpath=[abspath("scons/site_tools"), abspath("ext/dlr/scons-build-tools/site_tools")])
> ```
> 
> Why doesn't that fail?
> Maybe we can just wrap the strings in abspath?

`abspath()` normalizes the path separators to `\` on windows, so it basically includes the functionality of `windowsify`.

salkinium, 2019-05-05:
> I went with the repr way because theoretically a file path can contain arbitrary byte values except a few special cases (on linux for example path components can contain anything other than / and [...]

Ok, that is a very good reason, let's use that instead of raw strings!

> abspath() normalizes the path separators to \ on windows, so it basically includes the functionality of windowsify.

Ok, I'd prefer to use as much native Python functionality as possible, but that won't work with paths that get substituted later in SCons:
```
env.Append(MODM_OPENOCD_CONFIGFILES=["$BASEPATH/openocd.cfg"])
env.SetDefault(MODM_GDBINIT=["$BASEPATH/gdbinit"])
```
So there we would still need your solution.
Do you want to make those changes to the SCons module? I cannot really test it.

Sh4rK, 2019-05-05:
I can probably try it tomorrow.

Sh4rK, 2019-05-05:
I changed my mind about the raw strings vs. repr thing. Paths need to be represented outside of Python string literals as well (for example in the config files you mentioned) and I don't know how one would do that with random non-representable characters.

Also, a "correct" solution seems to be quite an undertaking (maybe rewriting everything using `pathlib`, as mentioned before). So for now I would just go with what you suggested first (windowsify + raw strings), to fix the immediate issue.

PR #204 (2019-05-05): [scons] Fix generated SConscript on Windows
I ended up with this solution after trying the raw strings version. The problem with that was that the generated `.gdbinit` file also needs to have double `\\`s, so I couldn't remove the doubling logic from `windowsify`.

This solves the immediate problem (at least for me), so I think it's worthwhile to merge this, and possibly come up with a cleaner solution later.

Fixes #192.
closed 2019-05-05 by salkinium