modm-io/modm #196: XMega support

TheSlat, 2019-04-24:
In an effort to learn lbuild, scons, etc I thought i would compile a project for my older boards. They are all XMega based. I noticed `xmega` was not in the filter list in repo.lb so I added it and ran into the following errors:

- spi/uart module instances are named with letters according the port they are on
- adc/dac signals are names with digits
- counters also have letters, `c0`, `c1`, `d0`, `d1` ...

the line in the spi (line 68) and uart (line 91) module:
`module.add_submodule(Instance(int(instance)))`
fails since the instance is named 'c' and 'd' on my atxmega128a4u

if I remove the int cast it progresses but fails with the message in the gpio module :
```
ERROR: These signal names are starting with a digit!
AcA::0, AcA::1, AcA::2, AcA::3, AcA::4, AcA::5, AcA::6, AcA::7, DacB::0, DacB::1
```

Snippet of the relevant XML from the device file:
```
      <gpio port="B" pin="0">
        <signal driver="ac" instance="b" name="0"/>
        <signal device-name="64" driver="adc" instance="a" name="7"/>
        <signal device-name="128" driver="adc" instance="a" name="8"/>
        <signal driver="adc" instance="a" name="refb0"/>
        <signal driver="adc" instance="b" name="0"/>
        <signal driver="adc" instance="b" name="refb0"/>
      </gpio>
```
```
    <driver name="spi" type="avr">
      <instance value="c"/>
      <instance value="d"/>
    </driver>
```
```
    <driver name="tc" type="tc">
      <instance value="c0"/>
      <instance value="c1"/>
      <instance value="d0"/>
      <instance value="d1"/>
      <instance value="e0"/>
      <instance device-pin="" value="exc"/>
      <instance device-pin="" value="resc"/>
      <instance device-pin="" value="resd"/>
      <instance device-pin="" value="rese"/>
    </driver>
    <driver name="twi" type="avr">
      <instance value="c"/>
      <instance value="e"/>
    </driver>
    <driver name="usart" type="avr">
      <instance value="c0"/>
      <instance value="c1"/>
      <instance value="d0"/>
      <instance value="d1"/>
      <instance value="e0"/>
    </driver>
```

salkinium, 2019-04-24:
I started to port our Xmega code from xpcc over, but then in the middle I realized that we're haven't used Xmega in a long time, and had absolutely no plans to use it again (AVRs are a bit annoying compared to STM32). So I kinda just stopped (since STM32 support was far more important), but I didn't want to delete the work I put in, so I just disabled the devices for it in `repo.lb` and kept the modules in it.

So Xmega isn't really supported in modm and I'm not even sure the modm-devices files are correct either. The Xmegas device files aren't even being generated in the CI anymore, I should probably delete them.

Maybe you want to fix this, but I don't know if you want to port both modm for both Xmega *and* SAM. If you do, then I'd start with verifying modm-devices, since the device files contain bad data like:
```
   <driver name="nvm" type="avr">
      <instance device-pin="" value=""/>
      <instance device-pin="" value="d_signatures"/>
      <instance device-pin="" value="e"/>
      <instance device-pin="" value="kbit"/>
    </driver>
```

TheSlat, 2019-04-24:
ok, just wondered if it was an easy fix. I defiantly don't want to port both right now, i just wanted to see a light blink today and play with lbuild/scons/modm. I found a really old board with an atmega2561 on it so I have something to work with. 🚨

closed 2019-06-16 by salkinium