modm-io/modm #201: Runtime configuration based Port/Pin assignments

Balint134, 2019-05-04:
Hi, I've been reading the documentation for quite a while now and cant really find an answer or example to my problem. I really like the idea of this library especially that it's really clean and seemingly easy to use although all the examples have the pin and port assignments set at compile time. My question would be; is that possible to use a dynamic configuration for pin assignments at runtime? 
My use case is basically I have different board manufacturers and they wire the same peripherals to different ports/pins so I'd need to have a way to use different pins and even different peripherals based on configuration.
For example, board `A` has a peripheral connected to the `SPI5` interface while board `B` uses SPI5 but on different pins while board `C` uses `SPI3` instead of 5.
Is there a way to solve this with modm or is the library really tailored around compile time assignments?
Thank you

salkinium, 2019-05-04:
You mean sonething like this?

```cpp
if (board_A) {
    SpiMaster5::connect<GpioA5::Mosi, GpioA6::Miso, GpioA7::Sck>();
    SpiMaster5::initialize<BoardA::SystemClock, 5_MHz>();
}
else if (board_B) {
    SpiMaster5::connect<GpioB5::Mosi, GpioB6::Miso, GpioB7::Sck>();
    SpiMaster5::initialize<BoardB::SystemClock, 5_MHz>();
}
else if (board_C) {
    SpiMaster3::connect<GpioC5::Mosi, GpioC6::Miso, GpioC7::Sck>();
    SpiMaster3::initialize<BoardC::SystemClock, 5_MHz>();
}
```

The connect methods compile into very compact code (just constants basically), including a few different configurations like this will probably still be less code than a fully dynamic computation.

salkinium, 2019-05-04:
You can of course just call `GpioA5::setAlternateFunction(id);`, which is what the connect function compiles down to, but you loose all of the compile-time sanity checks. The AF mapping is pretty random too for STM32.

Balint134, 2019-05-04:
Thanks for the response, yes something like that but in my case the number of combinations are close to 80 and I have at least 5-6 peripherals so the total number of permutations are very high. I was thinking more like
```
auto config = ConfigResolver::load()
Spi::connect(config.spi.interface, config.spi.mosi, config.spi.miso, config.spi.sck)
```
Apparently all the pin assignments are passed as template variables which is compile time only rather than runtime and I don't know if there's any way to get around this 🤔

salkinium, 2019-05-04:
There’s no runtime API for this I’m afraid. I believe mbed has such an API, but it’s completeness may be lacking.

However, you can take a look at the [modm:platform:gpio](https://github.com/modm-io/modm/blob/develop/src/modm/platform/gpio/stm32/module.lb) module to see how we transform the data from [modm-devices](https://github.com/modm-io/modm-devices/blob/develop/devices/stm32/stm32f4-05_07_15_17.xml#L241) into compile time structures, and perhaps transform it into a runtime friendly format instead. Or just generate one massive switch case with all permutations ;-p

salkinium, 2019-05-04:
You can even use modm-devices on it’s own and extend another HAL with that data. That was the whole point of separating modm (the *C++* library) from modm-devices after all.

Balint134, 2019-05-04:
Thanks for the tips, I'll take a look and see what can I do. Extending from `modm-devices` sounds interesting, wonder how complicated would that be 🤔

salkinium, 2019-05-04:
Here’s an introductory blog post: https://blog.salkinium.com/modm-devices/#using-device-files
Haven’t gotten around to writing more though.

salkinium, 2019-05-04:
I'm closing this since its out-of-scope for modm, but feel free to continue posting questions/issues here about using modm-devices.

closed 2019-05-04 by salkinium