modm-io/modm #208: Latest develop version with lbuild 1.11.2 does not work

Sh4rK, 2019-05-06:
I just upgraded my lbuild and pulled the latest develop changes, and this is what I get:

```
d:\Prog\modm\examples\nucleo_f429zi\blink>lbuild -r ..\..\..\repo.lb build
Recomputing device cache...
[WARNING] lbuild.repository: Namespacing repository filter 'windowsify' to 'modm.windowsify'!
[WARNING] lbuild.repository: Namespacing repository filter 'ord' to 'modm.ord'!
[WARNING] lbuild.repository: Namespacing repository filter 'chr' to 'modm.chr'!

ERROR: Cannot resolve name 'd:\Prog\modm\examples\nucleo_f429zi\blink\project.xml' of type Config!

Hint: Did you use the full name?
      Is the Config available and selected?
      Check your command line and config files.

Current project configuration:

ConfigNode(filename='command-line')

Parser(lbuild)
╰── Repository(modm @ ..\..\..)   modm: a barebone embedded library generator
    ├── EnumerationOption(target) = REQUIRED in [at90can128, at90can32, at90can64, at90pwm1, at90pwm161, at90pwm2, a ...
    ├── Configuration(modm:al-avreb-can)   # AL-AVREB_CAN Board
    ├── Configuration(modm:arduino-nano)   Arduino NANO
    ├── Configuration(modm:arduino-uno)   Arduino UNO
    ├── Configuration(modm:black-pill)   # Black Pill
    ├── Configuration(modm:blue-pill)   # Blue Pill
    ├── Configuration(modm:disco-f051r8)   # STM32F0DISCOVERY
    ├── Configuration(modm:disco-f072rb)   # STM32F072DISCOVERY
    ├── Configuration(modm:disco-f100rb)   # STM32VLDISCOVERY
    ├── Configuration(modm:disco-f303vc)   # STM32F3DISCOVERY
    ├── Configuration(modm:disco-f407vg)   # STM32F4DISCOVERY
    ├── Configuration(modm:disco-f429zi)   # STM32F429IDISCOVERY
    ├── Configuration(modm:disco-f469ni)   # STM32F469IDISCOVERY
    ├── Configuration(modm:disco-f746ng)   # STM32F7DISCOVERY
    ├── Configuration(modm:disco-f769ni)   # STM32F769IDISCOVERY
    ├── Configuration(modm:disco-l476vg)   # STM32L476DISCOVERY
    ├── Configuration(modm:nucleo-f031k6)   # NUCLEO-F031K6
    ├── Configuration(modm:nucleo-f042k6)   # NUCLEO-F042K6
    ├── Configuration(modm:nucleo-f103rb)   # NUCLEO-F103RB
    ├── Configuration(modm:nucleo-f303k8)   # NUCLEO-F303K8
    ├── Configuration(modm:nucleo-f401re)   # NUCLEO-F401RE
    ├── Configuration(modm:nucleo-f411re)   # NUCLEO-F411RE
    ├── Configuration(modm:nucleo-f429zi)   # NUCLEO-F429ZI
    ├── Configuration(modm:nucleo-g071rb)   # NUCLEO-G071RB
    ├── Configuration(modm:nucleo-l432kc)   # NUCLEO-L432KC
    ├── Configuration(modm:nucleo-l476rg)   # NUCLEO-L476RG
    ├── Configuration(modm:olimexino-stm32)   # Olimexino STM32
    ╰── Configuration(modm:stm32f030_demo)   # STM32F030 Demo Board
```

salkinium, 2019-05-06:
Can you try with: `pip3 install -U lbuild==1.11.1`?

salkinium, 2019-05-06:
Is this the standard `modm-io/modm/develop` branch?
Cos the warnings should be fixed now.

> [WARNING] lbuild.repository: Namespacing repository filter 'chr' to 'modm.chr'!

salkinium, 2019-05-06:
Oh no. I know what's happening here. WINDOWS. NOooooooo.
I use `:` as hierarchy separator for lbuild modules, and the `d:\` fucks this up.
Sorry, I'm fixing this with 1.11.3.

Sh4rK, 2019-05-06:
The warnings do seem to be fixed now, I don't remember what configuration of modm and lbuild I used in the first post.

salkinium, 2019-05-06:
Ok, I now first check if the config file exists and if it does not check if it contains `:` and if that's not the case, raise such a "not found" exception. Try v1.11.3.

Sh4rK, 2019-05-06:
Awesome, it works! Thanks!

closed 2019-05-06 by Sh4rK