modm-io/modm #212: UART 9 data bit mode on AVR

Sh4rK, 2019-05-11:
I'm developing a project for AVR where I need to use UART with 9 data bits. The current interface doesn't seem to support this. Do you think it would be reasonable (and not very complicated) to add support in modm for this, or should I just hack together something? :D

(I'm interfacing with [SeaTalk](http://www.thomasknauf.de/rap/seatalk1.htm), where the 9th data bit is used as the "command bit". Also, as this is a bus protocol, the RX receives what's sent on TX, and actually has to check whether the sent frame is correct before moving on, for collision detection.  I think this makes me unable to use the buffered UART functions, so it might make more sense to actually just copy the modm UART code and modify it for myself. Any thoughts?)

salkinium, 2019-05-11:
Yes, I think copying and modifying is your best choice here and gives you the most flexibility. Then you can also judge how much work it is to integrate into modm.

Sh4rK, 2019-05-11:
Okay, thanks!

closed 2019-05-11 by Sh4rK