modm-io/modm #216: Support for a CAN send buffer with priorities

ASMfreaK, 2019-05-14:
While reading through CAN driver's docs (a very lack thereof) and it's source code, I was wondering.
Is there a way to prioritize messages by their ID's with `CAN::sendMessage()`?
Should such prioritization happen?

Suppose, a device on CAN bus has two (or more) tasks. Each task send a heartbeat message with different ID (and therefore - priority) using `CAN::sendMessage()`. Suppose also this bus is a very busy one - with lots of other devices sending messages with higher (or equal) priority, than the highest priority of all taks on this device. 
Example: Two tasks, sending (1) `Msg(pr:0x10, len:8)` (2) `Msg(pr:0x20, len:1)`:
Task (2) occures **more** frequent, but has a **lower** priority.
Task (1) occures **less** frequent, but has a **higher** priority.
```
Msg(pr:0x20, len:1)
Msg(pr:0x20, len:1)
Msg(pr:0x20, len:1)
Msg(pr:0x20, len:1)
Msg(pr:0x20, len:1)
Msg(pr:0x20, len:1)
Msg(pr:0x10, len:8)
Msg(pr:0x10, len:8)
Msg(pr:0x20, len:1)
Msg(pr:0x20, len:1)
Msg(pr:0x20, len:1)
```

Is there a way to **ensure** that task (1) will always get it's message transmitted on time?

salkinium, 2019-05-14:
It's not implemented, you'd have to sort the queue every time you insert a message in it. With the right data structure (some sorted tree) that could still be efficient (even so, CAN is quite slow anyways).
The STM32 implementation also has a 3 message tx buffer in hardware, not sure if they prioritize in hardware too.

You can always just make it an option of the driver module, just like the buffer size is an option.

For our use-case we use CAN as a backend for communication (similar to UAVCAN) and the 29-bit message isn't really structured well for use of priority, so we just do a FIFO.

ASMfreaK, 2019-05-16:
According to [this](https://stackoverflow.com/questions/37077974/stm32-hal-can-tx-mailboxes), STMs TX mailboxes is indeed prioritized, but can be reconfigured otherwise:
(see sect. 24.7.1 Transmission handling in RM0008 Reference manual for STM32F10(1,2,3,5,7)xx)

> ## Transmit priority
> ### By identifier
> When more than one transmit mailbox is pending, the
transmission order is given by the identifier of the message
stored in the mailbox. The message with the lowest identifier
value has the highest priority according to the arbitration of the
CAN protocol. If the identifier values are equal, the lower mailbox
number will be scheduled first.
> ### By transmit request order
> The transmit mailboxes can be configured as a transmit FIFO by
setting the TXFP bit in the CAN_MCR register. In this mode the
priority order is given by the transmit request order. This mode is
very useful for segmented transmission.

salkinium, 2019-05-24:
You’re welcome to add this behavior, perhaps guarded by an lbuild option, since priority behavior may be implementation defined (AVRs and Hosted, Adapters, etc). We don‘t have a Priority Queue in modm, but the stdlib does: https://en.cppreference.com/w/cpp/container/priority_queue. That’s should suffice enough for now.

salkinium, 2019-09-09:
We plan to integrate UAVCAN support into modm, which seems to require this feature.

closed 2021-09-29 by salkinium