modm-io/modm #221: LPC memory layout / Shared RAM with on-chip API

strongly-typed, 2019-06-01:
In

https://github.com/modm-io/modm/blob/develop/tools/devices/lpc/lpc11c24.xml#L18
```xml
<driver name="core" type="cortex">
      <memory name="sram1" access="rwx" start="0x10000100" size="8"/>
```
the start of the RAM was assigned to to `0x1000 0100` with a size of 8 KiBytes.

Accordingly to Figure 6 in [UM10398.pdf](https://www.nxp.com/docs/en/user-guide/UM10398.pdf), RAM starts at `0x1000 0000`

[image]

Additionally, accordingly to section 17.4.1 ("Calling the C_CAN API"), "On-chip RAM from address 0x1000 0050 to 0x1000 00B8 is used by the CAN API.":

[image]

This might be the reason why the first used RAM was moved to `0x1000 0100` in xpcc back then.

I just want to clarify this to do the right things in the future. We probably need a feature in modm to reserve some RAM block for the use of on-chip peripherals / API code?!

salkinium, 2019-06-01:
> We probably need a feature in modm to reserve some RAM block for the use of on-chip peripherals / API code?!

That would be done with the linkerscript of the port, so modm already supports that. The device file should be as accurate to the hardware as possible with all interpretation being done in modm. So that start address should be fixed.

strongly-typed, 2019-06-02:
I will fix this into my modm port of LPC11C24.
How would the linkerscript reserve this RAM space?

salkinium, 2019-06-02:
I think the linkerscript would just start at 0x100 and ignore the first 256B. However you need to be careful about the vector table relocation which probably has an aligment requirement higher than that.

salkinium, 2019-06-02:
Oh, but this is a Cortex-M0, so there is no VTOR relocation anyways.

salkinium, 2021-03-29:
I'm closing this, since LPC is not supported by modm and probably won't be in the future either.

closed 2021-03-29 by salkinium