modm-io/modm #223: GpioInput and GpioOutput is aliased to Gpio

strongly-typed, 2019-06-01:
In [AVRs](https://github.com/modm-io/modm/blob/develop/src/modm/platform/gpio/at90_tiny_mega/pin.hpp.in#L27) and [STMs](https://github.com/modm-io/modm/blob/develop/src/modm/platform/gpio/stm32/pin.hpp.in#L30)
both `GpioOutput` and `GpioInput` are aliased to `Gpio`:
```
class Gpio{{ port ~ pin }};
using GpioOutput{{ port ~ pin }} = Gpio{{ port ~ pin }};
using GpioInput{{ port ~ pin }}  = Gpio{{ port ~ pin }};
```

I thought that it was a distinct feature of modm that once a pin was declared as as `GpioInput` it was not able to accidentally use the `setOutput` method on this pin.

Assume the case that a sinking/sourcing output of another device is directly connected to the pin of the microcontroller. To avoid hardware damage (short circuit), the pin is only used as `GpioInput` (e.g. by a board support package). Unintentionally `setOutput` et cetera were avoided at compile time.

Of course, on a micrcontroller without MMU, anybody could do anything at anytime, but I always liked checks in the code which hindered me to do stupid things.

Is it worth considering splitting `GpioInput` and `GpioOutput` again in separate classes and letting `Gpio` the only?

salkinium, 2019-06-02:
It is unclear how to map signals to input and output since some can be both. I aliased them so that in future this could be implemented correctly somehow. Unfortunately doing this right adds a lot of complexity to the template logic, maybe a generic C++ solution would be better.

salkinium, 2019-06-02:
I also remember a type-explosion issue that I didn’t want to deal with at the time related to signals, but looking at the code I cannot pin down what it was. There’s some preparation code like the `Gpio::Output`, `Gpio::Input` and `Gpio::IO` types that allow you to access the other interfaces from the same pin if you really need to.

salkinium, 2021-06-16:
I decided not to fix this.

closed 2021-06-16 by salkinium