modm-io/modm #238: Support for chinese STM32F103C8T6 clones

ASMfreaK, 2019-07-09:
Hi!
We were unlucky to buy a bunch of BluePills with chinese clone of STM32 called CKS32F103C8T6. As far as I understand, it is a perfect replacement, but openocd fails to upload a firmware in it. If I understand correctly, it is not perfectly compatible - openocd recieves the same CPUTAPID as STM32F3x series MCUs. If I put 
```openocd
set CPUTAPID 0x2ba01477
```
just before include of mcu's config (so it cannot be fixed by using a custom script), it uploads just fine and works.
It requires futher testing.
I want to propose a change to add an option to bluepill bdf's to  support it.

ASMfreaK, 2019-07-09:
I don't have a ready-made pr for this, so I want an advice for a best way to implement this. Should it be a new target?

salkinium, 2019-07-10:
I think this is out-of-scope for modm. There is a risk of "customization creep" where a lot of lbuild options are added to modm. Instead our vision was to take the generated modm output and modify it and then commit those changes.

Piping the OpenOCD config options through lbuild were mostly a hack without a good concept. The blue pill config inside the repository has the programmer hard-coded (#234), which is obviously an issue, requiring yet another lbuild option (#235).
It would be better if you could overwrite the default openocd config file  entirely and replace it with your own. I have to think about how that would work.

PR #240 (2019-07-11): [openocd] Do not hardcode user options in built-in configs
The modm board configurations hard-coded the programmer, which is an issue when using a different one.

This replaces #235 and fixes #234 and fixes #238.

cc @ASMfreaK
closed 2019-07-11 by salkinium