modm-io/modm #243: STM32 acknowledge of interrupt clears all pending bits

19joho66, 2019-07-15:
I've used the STM32 feature, where several external interrupts are handled by one ISR (EXTI9_5). In my application I used GPIO5 to GPIO7. When several interrupt request are pending an I acknowlede just one of them, the other ones are cleared too.

The acknowledge routine looks like this:

`inline static void acknowledgeExternalInterruptFlag() { EXTI->PR |= mask; }`

It reads the content of the PR register, ORs it with the mask and writes the result back to the PR register.  But writing a 1 to the PR register will reset the according bit and therefore all bits are reset.

I've changed the routine to:

`inline static void acknowledgeExternalInterruptFlag() { EXTI->PR = mask; } `

and now it seems to work correctly. 

Maybe there is someone who can confirm this behaviour.

salkinium, 2019-07-15:
According to the manual, the bits in that register all have `rc_w1` behavior:
[image]
[image]

Clearly a read-modify-write is not required to clear this bit, a 1-write is enough, *and* for you it doesn't actually do the right thing anyways. Do you want to open a PR with your fix?

salkinium, 2019-07-15:
What target are you using? The above is from the STM32F4 reference manual, but there are only two (major) implementations of GPIO in STM32 anyways.

19joho66, 2019-07-15:
I'm using a STM32F3Discovery board (STM32F303 processor). 
The problem with `inline static void acknowledgeExternalInterruptFlag() { EXTI->PR |= mask; }`
is, that PR is read into a cpu-register which will then be ORed with the mask (so it does basicaly nothing) and then  the  register value is written back to PR which then clears all set interrupt pending bits (and not just the desired one).

I can do an PR propably tomorrow.

salkinium, 2019-07-15:
> then the register value is written back to PR which then clears all set interrupt pending bits (and not just the desired one).

Oh, I understand, of course the PR register contains ALL pending bits, so this then always clears ALL pending bits. 🤦‍♂️

PR #244 (2019-07-16): Bugfix: STM32 acknowledge of interrupt clears all pending bits 
Fixes #243:

When using `inline static void acknowledgeExternalInterruptFlag() { EXTI->PR |= mask; }`,
EXTI->PR is read into a cpu-register which will then be ORed with the mask (so it does basicaly nothing) and then the register value is written back to PR which then clears all set interrupt pending bits (and not just the desired one).
closed 2019-07-16 by salkinium