modm-io/modm #245: CrashCatcher hard fault handler asm file not linked with cmake

XDjackieXD, 2019-07-16:
When compiling using CMake, the CrashCatcher_armv7m.S doesn't get assembled and linked into the resulting binary.
When using Scons, everything is working as expected (but it doesn't integrate with CLion which I'm using as my IDE)

salkinium, 2019-07-16:
Can confirm, the `.S` file is added to the source list, but not compiled:

```cmake
SET(SOURCE_FILES
    modm/ext/crashcatcher/src/CrashCatcher.c
    modm/ext/crashcatcher/src/CrashCatcher_armv7m.S
    modm/ext/tlsf/tlsf.c
    ...
    modm/src/modm/platform/core/clock.cpp
    modm/src/modm/platform/core/cxxabi.cpp
    modm/src/modm/platform/core/default_handler.sx
    modm/src/modm/platform/core/delay.cpp
    modm/src/modm/platform/core/startup.c
    ...
    ${APP_SOURCE_FILES}
)
```

The compiled files are these:
```
 $ make build-release
Scanning dependencies of target hard_fault
[  3%] Building C object CMakeFiles/hard_fault.dir/modm/ext/crashcatcher/src/CrashCatcher.c.o
[  5%] Building C object CMakeFiles/hard_fault.dir/modm/ext/tlsf/tlsf.c.o
...
[ 38%] Building CXX object CMakeFiles/hard_fault.dir/modm/src/modm/platform/core/clock.cpp.o
[ 40%] Building CXX object CMakeFiles/hard_fault.dir/modm/src/modm/platform/core/cxxabi.cpp.o
[ 42%] Building CXX object CMakeFiles/hard_fault.dir/modm/src/modm/platform/core/default_handler.sx.o
[ 44%] Building CXX object CMakeFiles/hard_fault.dir/modm/src/modm/platform/core/delay.cpp.o
[ 49%] Building C object CMakeFiles/hard_fault.dir/modm/src/modm/platform/core/startup.c.o
...
[ 98%] Building CXX object CMakeFiles/hard_fault.dir/main.cpp.o
[100%] Linking CXX executable hard_fault.elf
   text	   data	    bss	    dec	    hex	filename
   7404	    552	17170312	17178268	1061e9c	hard_fault.elf
```

So the `.sx` file gets compiled, but the `.S` file does not?!? wat.

salkinium, 2019-07-16:
Ah, it's because of these lines in the [`CMakeLists.txt.in`](https://github.com/modm-io/modm/blob/develop/tools/build_script_generator/cmake/resources/CMakeLists.txt.in#L11-L12):

```cmake
FILE(GLOB_RECURSE APP_STARTUP *.sx)
SET_SOURCE_FILES_PROPERTIES(${APP_STARTUP} PROPERTIES LANGUAGE CXX)
```

Adding `*.S` to them makes it work:

```cmake
FILE(GLOB_RECURSE APP_STARTUP *.sx *.S)
SET_SOURCE_FILES_PROPERTIES(${APP_STARTUP} PROPERTIES LANGUAGE CXX)
```

The CMake generator needs to be refactored, it does not have the same features as the SCons generator.

salkinium, 2019-07-16:
I'm not very good at CMake, but I realize that it's the defacto standard for IDE etc integration, so modm needs to support it. I'd appreciate some help by someone who knows what their doing ;-P

closed 2019-07-23 by salkinium