modm-io/modm #246: [regression] OpenOCD fails to find cmake-built executable

ASMfreaK, 2019-07-16:
There is a bug, introduced by #236 and, in part, by #240. 
When uploading with OpenOCD, this substitution, generated by lbuild, is producing wrong elf path. See [here](https://github.com/modm-io/modm/blob/develop/tools/build_script_generator/module.lb#L147)
```python
    elf_files = {
        "release": env.relative_outpath(env.collector_values("elf.release")[0], "modm/"),
        "debug": env.relative_outpath(env.collector_values("elf.debug")[0], "modm/"),
    }
```

this should be 

```python
    elf_files = {
        "release": env.relative_outpath(env.collector_values("elf.release")[0], ""),
        "debug": env.relative_outpath(env.collector_values("elf.debug")[0], ""),
    }
```

ASMfreaK, 2019-07-16:
I've updated my comment

ASMfreaK, 2019-07-16:
Should I open PR for this?

salkinium, 2019-07-16:
Ah damn, I forgot that OpenOCD uses relative paths to the *call site*, instead of the actual configuration… The `scons program` command supplies the source explicitly, so this never gets called.
```py
env.Alias("program", env.OpenOcd(program, commands=["modm_program $SOURCE"]))
```

> Should I open PR for this?

Yes, but I'd prefer a way to make OpenOCD resolve the path relative to the config file, not relative to the CWD. I cannot guarantee that in all the build system wrapping the CWD is preserved (especially when calling from an IDE).

ASMfreaK, 2019-07-16:
Maybe we should make cmake makefiles behave the same way as scons?

salkinium, 2019-07-16:
Yes, then we would not need the hardcoded file paths in the config in the first place!

Only the `:build:cmake` Makefile uses these commands explicitly, the GDB is passed the source explicitly, so is everything in `:build:scons`.

ASMfreaK, 2019-07-16:
As far as I understood, googling for this, it is nearly impossible. Maybe we can do something like:
```
proc program_release {} {
        modm_program $BUILD_DIR/build/cmake-build-release/project_name.elf
}
proc program_debug {} {
        modm_program $BUILD_DIR/build/cmake-build-debug/project_name.elf
}
```
Makefile:
```makefile
upload-release: build-release
        @openocd -c "set BUILD_DIR $(BUILD_DIR)" -f modm/openocd.cfg -c "program_release"
```

ASMfreaK, 2019-07-16:
> Yes, then we would not need the hardcoded file paths in the config in the first place!
> 
> Only the `:build:cmake` Makefile uses these commands explicitly, the GDB is passed the source explicitly, so is everything in `:build:scons`.

Let's fix it by explicitly calling with the executable?

salkinium, 2019-07-16:
> Let's fix it by explicitly calling with the executable?

Yes, it's just these two lines:

```
upload-debug: build-debug
	@openocd -f modm/openocd.cfg -c "program_debug"

upload-release: build-release
	@openocd -f modm/openocd.cfg -c "program_release"
```

ASMfreaK, 2019-07-16:
Maybe we should also remove `program_release` and `program_debug`  `proc`s ?

salkinium, 2019-07-16:
Yes, and there is also another file thing in `openocd_gdbinit`, which is also explicitly passed and never used.

```
%% for profile, path in elf_files.items()
define file_{{profile}}
    file
    file {{ path | modm.windowsify(escape_level=1) }}
end
%% endfor
```

ASMfreaK, 2019-07-16:
Working on a PR for this

PR #247 (2019-07-16): Fix openocd makefile
Fixes #246 
@salkinium Is there anything other that needs to be excluded?
closed 2019-07-16 by salkinium