modm-io/modm #251: pip install fails with error about colorful

strongly-typed, 2019-07-23:
From the [docs](https://github.com/modm-io/modm/blame/develop/docs/src/guide/installation.md#L63):

```
➜  ~ docker run -it python:latest /bin/bash
root@40cf1379b075:/# pip3 install --user modm
[...]
ERROR: lbuild 1.12.2 has requirement colorful==0.4.4, but you'll have colorful 0.5.0 which is incompatible.
```

It works if you first install lbuild and then modm:
```
➜  ~ docker run -it python:latest /bin/bash
root@15cae4a90072:/# pip3 install --user --quiet lbuild
  WARNING: The script lbuild is installed in '/root/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
WARNING: You are using pip version 19.1.1, however version 19.2.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
root@15cae4a90072:/# pip3 install --user --quiet modm
  WARNING: The script modm is installed in '/root/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
WARNING: You are using pip version 19.1.1, however version 19.2.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
root@15cae4a90072:/#
```

It seems to work with Ubuntu:
```
➜  ~ docker run -it ubuntu:latest /bin/bash
```

What's wrong here?

Full log:
```
➜  ~ docker run -it python:latest /bin/bash
root@40cf1379b075:/# pip3 install --user modm
Collecting modm
  Downloading https://files.pythonhosted.org/packages/93/9b/794ebf84745d0444c04310bd3a1e113db00edef6d6c6208a2a42b2ec6d1e/modm-0.1.0-py3-none-any.whl
Collecting lxml (from modm)
  Downloading https://files.pythonhosted.org/packages/ef/7a/a42b825b27eaa0deedec913e797dd3e4bef51f21d5f0027a87562342fd25/lxml-4.3.4-cp37-cp37m-manylinux1_x86_64.whl (5.7MB)
     |████████████████████████████████| 5.7MB 3.1MB/s
Collecting lbuild (from modm)
  Downloading https://files.pythonhosted.org/packages/6b/9c/b41e4bfba946536c45b196323254213a40b5cc7deaef23958cad3063c1b0/lbuild-1.12.2-py3-none-any.whl (68kB)
     |████████████████████████████████| 71kB 4.2MB/s
Collecting colorful (from modm)
  Downloading https://files.pythonhosted.org/packages/a1/d7/59181df2dadf1234afddc93f1b7d965e1614423bb6a87845ea49f841c88a/colorful-0.5.0-py2.py3-none-any.whl (201kB)
     |████████████████████████████████| 204kB 3.1MB/s
Collecting pyelftools (from modm)
  Downloading https://files.pythonhosted.org/packages/fa/9a/0674cb1725196568bdbca98304f2efb17368b57af1a4bb3fc772c026f474/pyelftools-0.25.tar.gz (499kB)
     |████████████████████████████████| 501kB 3.5MB/s
Collecting pathlib (from modm)
  Downloading https://files.pythonhosted.org/packages/ac/aa/9b065a76b9af472437a0059f77e8f962fe350438b927cb80184c32f075eb/pathlib-1.0.1.tar.gz (49kB)
     |████████████████████████████████| 51kB 3.2MB/s
Collecting gitpython>=2.1.11 (from lbuild->modm)
  Downloading https://files.pythonhosted.org/packages/1a/44/fc1ce6e0692f09d1bcdbbb5d2bd008acd824505a12dd85a627ef74f44844/GitPython-2.1.12-py2.py3-none-any.whl (452kB)
     |████████████████████████████████| 460kB 2.9MB/s
Collecting jinja2 (from lbuild->modm)
  Downloading https://files.pythonhosted.org/packages/1d/e7/fd8b501e7a6dfe492a433deb7b9d833d39ca74916fa8bc63dd1a4947a671/Jinja2-2.10.1-py2.py3-none-any.whl (124kB)
     |████████████████████████████████| 133kB 3.4MB/s
Collecting anytree>=2.4.3 (from lbuild->modm)
  Downloading https://files.pythonhosted.org/packages/78/bf/9300ecef72d105b3a76de6930916458d6bb8c7e787f5efb1c510bf898873/anytree-2.6.0.tar.gz
Collecting ddt>=1.1.1 (from gitpython>=2.1.11->lbuild->modm)
  Downloading https://files.pythonhosted.org/packages/cf/f5/f83dea32dc3fb3be1e5afab8438dce73ed587740a2a061ae2ea56e04a36d/ddt-1.2.1-py2.py3-none-any.whl
Collecting gitdb>=0.6.4 (from gitpython>=2.1.11->lbuild->modm)
  Downloading https://files.pythonhosted.org/packages/e3/95/7e5d7261feb46c0539ac5e451be340ddd64d78c5118f2d893b052c76fe8c/gitdb-0.6.4.tar.gz (400kB)
     |████████████████████████████████| 409kB 3.2MB/s
Collecting gitdb2>=2.0.0 (from gitpython>=2.1.11->lbuild->modm)
  Downloading https://files.pythonhosted.org/packages/da/30/a407568aa8d8f25db817cf50121a958722f3fc5f87e3a6fba1f40c0633e3/gitdb2-2.0.5-py2.py3-none-any.whl (62kB)
     |████████████████████████████████| 71kB 2.9MB/s
Collecting MarkupSafe>=0.23 (from jinja2->lbuild->modm)
[... 654 lines cut ...]
    100% |################################| 71kB 3.1MB/s
Collecting ddt>=1.1.1 (from gitpython>=2.1.11->lbuild->modm)
  Downloading https://files.pythonhosted.org/packages/cf/f5/f83dea32dc3fb3be1e5afab8438dce73ed587740a2a061ae2ea56e04a36d/ddt-1.2.1-py2.py3-none-any.whl
Collecting gitdb>=0.6.4 (from gitpython>=2.1.11->lbuild->modm)
  Downloading https://files.pythonhosted.org/packages/e3/95/7e5d7261feb46c0539ac5e451be340ddd64d78c5118f2d893b052c76fe8c/gitdb-0.6.4.tar.gz (400kB)
    100% |################################| 409kB 1.8MB/s
Collecting smmap2>=2.0.0 (from gitdb2>=2.0.0->gitpython>=2.1.11->lbuild->modm)
  Downloading https://files.pythonhosted.org/packages/55/d2/866d45e3a121ee15a1dc013824d58072fd5c7799c9c34d01378eb262ca8f/smmap2-2.0.5-py2.py3-none-any.whl
Collecting smmap>=0.8.5 (from gitdb>=0.6.4->gitpython>=2.1.11->lbuild->modm)
  Downloading https://files.pythonhosted.org/packages/bc/aa/b744b3761fff1b10579df996a2d2e87f124ae07b8336e37edc89cc502f86/smmap-0.9.0.tar.gz
Building wheels for collected packages: pyelftools, pathlib, anytree, gitdb, smmap
  Running setup.py bdist_wheel for pyelftools ... done
  Stored in directory: /root/.cache/pip/wheels/55/2f/15/4ce6885a52f475de68c16f3116a94d2156d588390cdb6c507c
  Running setup.py bdist_wheel for pathlib ... done
  Stored in directory: /root/.cache/pip/wheels/f9/b2/4a/68efdfe5093638a9918bd1bb734af625526e849487200aa171
  Running setup.py bdist_wheel for anytree ... done
  Stored in directory: /root/.cache/pip/wheels/6b/e2/88/ffb3e712f4e961427fd2dab733bf86dcd725d35b659fc18072
  Running setup.py bdist_wheel for gitdb ... done
  Stored in directory: /root/.cache/pip/wheels/d1/7e/c5/1093bc36622d7d06ed6520a50280771ebff8c190d92bd402b6
  Running setup.py bdist_wheel for smmap ... done
  Stored in directory: /root/.cache/pip/wheels/d6/a5/09/cb584a95ad332da2bf0265a8b0423c562a99553a3edf148d12
Successfully built pyelftools pathlib anytree gitdb smmap
Installing collected packages: pyelftools, colorful, MarkupSafe, jinja2, anytree, smmap2, gitdb2, ddt, smmap, gitdb, gitpython, lxml, lbuild, pathlib, modm
Successfully installed MarkupSafe-1.1.1 anytree-2.6.0 colorful-0.5.0 ddt-1.2.1 gitdb-0.6.4 gitdb2-2.0.5 gitpython-2.1.12 jinja2-2.10.1 lbuild-1.12.2 lxml-4.3.4 modm-0.1.0 pathlib-1.0.1 pyelftools-0.25 smmap-0.9.0 smmap2-2.0.5
```

salkinium, 2019-07-23:
[colorful](https://github.com/timofurrer/colorful) was releasing breaking API changes as minor version patches, which broke lbuild a number of times… I stopped trusting the maintainer and thus hard froze the version to 0.4.4.
SInce I'm only using boldness, let me see if I can just remove the colorful package entirely.

ASMfreaK, 2019-07-23:
Maybe you want to use plumbum as app-driver? (shameless plug, as I wrote, in part. localization for it)

salkinium, 2019-07-23:
lbuild 1.12.3 has no dependency on colorful anymore, I've replaced it with raw ANSI escape codes. lbuild only uses bold, underline and red.

salkinium, 2019-07-23:
> Maybe you want to use plumbum as app-driver?

Nah, I wanted to have a rich color output for lbuild, but during implementation I realized it's too noisy and looks inconsistent with different shell backgrounds and now just use bold and underline. Red only for REQUIRED option values.

strongly-typed, 2019-07-23:
I can confirm that
```
➜  ~ docker run -it python:latest /bin/bash
pip3 install --user modm
```
now works with `lbuild-1.12.3`.

Which CI step shall we add to check for this dependency error in the future?

strongly-typed, 2019-07-23:
And `modm` still installs `colorful`. Is this correct?

salkinium, 2019-07-23:
> And modm still installs colorful. Is this correct?

Oh, good point, let me update that.

strongly-typed, 2019-07-23:
The link to the homepage [here](https://pypi.org/project/modm/) is dead.

salkinium, 2019-07-23:
I've updated the modm package to v0.1.1 to remove the colorful package and update the homepage.

strongly-typed, 2019-07-23:
Looks good to me.

closed 2019-07-23 by strongly-typed