modm-io/modm #256: warning: implicitly-declared ... GenericTimestamp ... is deprecated [-Wdeprecated-copy]

strongly-typed, 2019-07-25:
```
In file included from modm/src/modm/processing/timer/timeout.hpp:142,
                 from modm/src/modm/processing/timer.hpp:15,
                 from main.cpp:16:
modm/src/modm/processing/timer/timeout_impl.hpp: In instantiation of 'void modm::GenericTimeout<Clock, TimestampType>::restart(TimestampType) [with Clock = modm::Clock; TimestampType = modm::GenericTimestamp<unsigned int>]':
main.cpp:50:23:   required from here
modm/src/modm/processing/timer/timeout_impl.hpp:35:10: warning: implicitly-declared 'constexpr modm::GenericTimestamp<unsigned int>& modm::GenericTimestamp<unsigned int>::operator=(const modm::GenericTimestamp<unsigned int>&)' is deprecated [-Wdeprecated-copy]
   35 |  endTime = Clock::template now<TimestampType>() + time;
      |  ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from modm/src/modm/architecture/interface/clock.hpp:20,
                 from modm/src/modm/platform/clock/common.hpp:17,
                 from modm/src/modm/platform.hpp:18,
                 from main.cpp:14:
modm/src/modm/processing/timer/timestamp.hpp:49:2: note: because 'modm::GenericTimestamp<unsigned int>' has user-provided 'modm::GenericTimestamp<T>::GenericTimestamp(const modm::GenericTimestamp<T>&) [with T = unsigned int]'
   49 |  GenericTimestamp(const GenericTimestamp<T> &other) :
      |  ^~~~~~~~~~~~~~~~
```

salkinium, 2019-07-25:
What target/example are you compiling, what compiler are you using?

strongly-typed, 2019-07-25:
```
Building: avr/can/mcp2515_uart with SCons
```

```
➜  ~ avr-gcc --version
avr-gcc (GCC) 9.1.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
as from the install instructions from brew

```
➜  ~ brew info avr-gcc
osx-cross/avr/avr-gcc: stable 9.1.0 (bottled), HEAD
GNU compiler collection for AVR 8-bit and 32-bit Microcontrollers
https://www.gnu.org/software/gcc/gcc.html
/usr/local/Cellar/avr-gcc/9.1.0 (1,728 files, 204.9MB) *
```

On CircleCI, we use
```
/work/avr-gcc/avr-gcc/bin/avr-g++
avr-g++ (GCC) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
which is magically downloaded from your webserver?!

salkinium, 2019-07-25:
> which is magically downloaded from your webserver?!

Ugh, because Ubuntu doesn't have an up-to-date avr-gcc:
https://github.com/salkinium/docker-avr-gcc-7

salkinium, 2019-07-25:
AVR GCC 9 has better diagnostics, and these warnings also apply to ARM GCC 9 too.

salkinium, 2019-07-25:
I'm checking to see if I can build avr-gcc 9.1.0 and rename `docker-avr-gcc-7` to `docker-avr-gcc`.

salkinium, 2019-07-25:
New avr-gcc is available here:
https://github.com/salkinium/docker-avr-gcc/releases/tag/v9.1.0

Updating the Docker image now.

salkinium, 2019-07-25:
Now also shown in the CI:
https://circleci.com/gh/modm-io/modm/13238

salkinium, 2019-07-30:
FYI: There's a partial fix available here: https://github.com/salkinium/modm/commit/302a8f9fcb7a8405cdcf28c722cc3e8fe45b8408

But I still need to understand what the rules for default copy-constructors are and like to fix the other warnings too in one go.

salkinium, 2019-08-01:
Fixed

closed 2019-08-01 by salkinium