modm-io/modm #257: modm::sab warning: cast between incompatible pointer to member types from

strongly-typed, 2019-07-25:
```
In file included from main.cpp:21:
modm/src/modm/communication/sab/slave.hpp:275:59: warning: cast between incompatible pointer to member types from 'void (InOut::*)(modm::sab::Response&)' to 'modm::sab::Action::Callback' {aka 'void (modm::sab::Callable::*)(modm::sab::Response&, const void*)'} [-Wcast-function-type]
  275 |    reinterpret_cast<modm::sab::Action::Callback>(&function), \
      |                                                           ^
main.cpp:87:2: note: in expansion of macro 'SAB_ACTION'
   87 |  SAB_ACTION( 'I', inOut,   InOut::readInput,   0 ),
      |  ^~~~~~~~~~
```

strongly-typed, 2019-07-25:
https://circleci.com/gh/modm-io/modm/13126?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link

salkinium, 2019-07-25:
That's not new. I'll be using SAB in the ELVA project so I'll property fix it then.

closed 2026-10-31 by salkinium