modm-io/modm #263: cmake not idempotent

strongly-typed, 2019-07-30:
Calling `cmake` on the first run created different result than on the preceding runs.

```
➜  build git:(feature/ci_macos) ✗ pwd
/Users/usr/Dev/modm/modm-strongly-ci-macos/examples/arduino_uno/basic/digital_read_serial/build
➜  build git:(feature/ci_macos) ✗ cmake ..
-- The C compiler identification is AppleClang 10.0.1.10010046
-- The CXX compiler identification is AppleClang 10.0.1.10010046
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2")
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/usr/Dev/modm/modm-strongly-ci-macos/examples/arduino_uno/basic/digital_read_serial/build
➜  build git:(feature/ci_macos) ✗ cmake ..
-- Configuring done
You have changed variables that require your cache to be deleted.
Configure will be re-run and you may have to reset some variables.
The following variables have changed:
CMAKE_C_COMPILER= avr-gcc
CMAKE_CXX_COMPILER= avr-g++

-- The C compiler identification is GNU 9.1.0
-- The CXX compiler identification is GNU 9.1.0
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - no
-- Check for working C compiler: /usr/local/bin/avr-gcc
-- Check for working C compiler: /usr/local/bin/avr-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking whether CXX compiler has -isysroot
-- Checking whether CXX compiler has -isysroot - yes
-- Checking whether CXX compiler supports OSX deployment target flag
-- Checking whether CXX compiler supports OSX deployment target flag - no
-- Check for working CXX compiler: /usr/local/bin/avr-g++
-- Check for working CXX compiler: /usr/local/bin/avr-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2")
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/usr/Dev/modm/modm-strongly-ci-macos/examples/arduino_uno/basic/digital_read_serial/build
➜  build git:(feature/ci_macos) ✗ cmake ..
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/usr/Dev/modm/modm-strongly-ci-macos/examples/arduino_uno/basic/digital_read_serial/build
```

strongly-typed, 2019-07-30:
That might be a reason why [TravisCI](https://api.travis-ci.com/v3/job/220521690/log.txt) fails with `avr-g++: error: unrecognized command line option '-mmacosx-version-min=10.14'`

Related to #261

salkinium, 2019-07-30:
Yeah, noticed that too, I need some help with CMake, I clearly didn’t do it right.

salkinium, 2019-07-30:
Maybe turning of the caching in `configure-gcc.cmake` will fix this?

strongly-typed, 2019-07-30:
I discovered, that the generated `configure-gcc.cmake` on my machine is missing the [`hosted` part](https://github.com/modm-io/modm/blob/develop/tools/build_script_generator/cmake/cmake_scripts/configure-gcc.cmake.in#L30) from the template.

strongly-typed, 2019-07-30:
> I discovered, that the generated `configure-gcc.cmake` on my machine is missing the [`hosted` part](https://github.com/modm-io/modm/blob/develop/tools/build_script_generator/cmake/cmake_scripts/configure-gcc.cmake.in#L30) from the template.

... which is totally ok, we are building for `avr` 🙃

strongly-typed, 2019-07-30:
We (@FelixStau) think that swapping `include` and `project`
```
include(./modm/repo.cmake)
project(digital_read_serial)
```
solves the problem. If nothing is set, `project` defines (already) the default languages `C` and `C++`.

closed 2019-07-31 by salkinium