modm-io/modm #269: Auto-generated CMake - Ubuntu 18.04

nicoBruhn, 2019-08-19:
The auto generated cmake no longer compiles on ubuntu 18.04. The output from the build script on my computer can be seen here: https://justpaste.it/2wmyo

The auto generated cmake script work with commit: 20b145787ee8e695ad7dfd30ee4bb92d7aebcc9a

salkinium, 2019-08-19:
Hm, so it detects the compiler correctly, but doesn't set the flags right? That's very weird since our CI runs on Ubuntu 18.04, but I probably messed something up during refactoring. Can you show me the generated `modm/repo.cmake` as well as `modm/cmake/configure-gcc.cmake`?

nicoBruhn, 2019-08-19:
configure.cmake: https://justpaste.it/35735
repo.cmake: https://justpaste.it/49sg2

salkinium, 2019-08-19:
I don't see any obviously wrong things in there, and [this PR](https://github.com/modm-io/modm/pull/264/files) was merged before the commit you pointed out. I'm not sure why this doesn't work.

nicoBruhn, 2019-08-19:
I think the issue started further back. At commit 8c199919f1d241d82577cf01e106f532ae166ead
cmake detects the wrong compiler.

nicoBruhn, 2019-08-19:
I found a fix for it. I am not sure exactly why the code is not working.
In repo.cmake replace
`
string(APPEND CCFLAGS "${ARCHFLAGS}")
string(APPEND ASFLAGS "${ARCHFLAGS}")
string(APPEND LINKFLAGS "${ARCHFLAGS}")
string(APPEND CMAKE_C_FLAGS         "${CCFLAGS} ${CFLAGS}")
string(APPEND CMAKE_C_FLAGS_RELEASE "${CCFLAGS_RELEASE}")
string(APPEND CMAKE_C_FLAGS_DEBUG   "${CCFLAGS_DEBUG}")

string(APPEND CMAKE_CXX_FLAGS         "${CCFLAGS} ${CXXFLAGS}")
string(APPEND CMAKE_CXX_FLAGS_RELEASE "${CCFLAGS_RELEASE}")
string(APPEND CMAKE_CXX_FLAGS_DEBUG   "${CCFLAGS_DEBUG}")

string(APPEND CMAKE_ASM_FLAGS         "${ASFLAGS}")
string(APPEND CMAKE_ASM_FLAGS_RELEASE "${ASFLAGS_RELEASE}")
string(APPEND CMAKE_ASM_FLAGS_DEBUG   "${ASFLAGS_DEBUG}")

string(APPEND CMAKE_EXE_LINKER_FLAGS         "${LINKFLAGS}")
string(APPEND CMAKE_EXE_LINKER_FLAGS_RELEASE "${LINKFLAGS_RELEASE}")
string(APPEND CMAKE_EXE_LINKER_FLAGS_DEBUG   "${LINKFLAGS_DEBUG}")
`
with
`
SET(CMAKE_C_FLAGS "${ARCHFLAGS} ${CCFLAGS} ${CFLAGS}" CACHE INTERNAL "c compiler flags")
SET(CMAKE_C_FLAGS_RELEASE   "${CCFLAGS_RELEASE}"      CACHE INTERNAL "c compiler flags release")
SET(CMAKE_C_FLAGS_DEBUG     "${CCFLAGS_DEBUG}"        CACHE INTERNAL "c compiler flags debug")

SET(CMAKE_CXX_FLAGS "${ARCHFLAGS} ${CCFLAGS} ${CXXFLAGS}" CACHE INTERNAL "cxx compiler flags")
SET(CMAKE_CXX_FLAGS_RELEASE "${CCFLAGS_RELEASE}"          CACHE INTERNAL "cxx compiler flags release")
SET(CMAKE_CXX_FLAGS_DEBUG   "${CCFLAGS_DEBUG}"            CACHE INTERNAL "cxx compiler flags debug")

SET(CMAKE_ASM_FLAGS "${ARCHFLAGS} ${ASFLAGS}" CACHE INTERNAL "asm compiler flags")
SET(CMAKE_ASM_FLAGS_RELEASE "${ASFLAGS_RELEASE}")
SET(CMAKE_ASM_FLAGS_DEBUG   "${ASFLAGS_DEBUG}")

SET(CMAKE_EXE_LINKER_FLAGS  "${ARCHFLAGS} ${LINKFLAGS}" CACHE INTERNAL "linker flags")
SET(CMAKE_EXE_LINKER_FLAGS_RELEASE "${LINKFLAGS_RELEASE}")
SET(CMAKE_EXE_LINKER_FLAGS_DEBUG   "${LINKFLAGS_DEBUG}")
`

nicoBruhn, 2019-08-19:
I think the issue is that the cmake compiler flags are set when the project() command i called and therefore overwritten. A solution to this problem is replacing the following code:
```
string(APPEND CCFLAGS "${ARCHFLAGS}")
string(APPEND ASFLAGS "${ARCHFLAGS}")
string(APPEND LINKFLAGS "${ARCHFLAGS}")
string(APPEND CMAKE_C_FLAGS         "${CCFLAGS} ${CFLAGS}")
string(APPEND CMAKE_C_FLAGS_RELEASE "${CCFLAGS_RELEASE}")
string(APPEND CMAKE_C_FLAGS_DEBUG   "${CCFLAGS_DEBUG}")
string(APPEND CMAKE_CXX_FLAGS         "${CCFLAGS} ${CXXFLAGS}")
string(APPEND CMAKE_CXX_FLAGS_RELEASE "${CCFLAGS_RELEASE}")
string(APPEND CMAKE_CXX_FLAGS_DEBUG   "${CCFLAGS_DEBUG}")

string(APPEND CMAKE_ASM_FLAGS         "${ASFLAGS}")
string(APPEND CMAKE_ASM_FLAGS_RELEASE "${ASFLAGS_RELEASE}")
string(APPEND CMAKE_ASM_FLAGS_DEBUG   "${ASFLAGS_DEBUG}")

string(APPEND CMAKE_EXE_LINKER_FLAGS         "${LINKFLAGS}")
string(APPEND CMAKE_EXE_LINKER_FLAGS_RELEASE "${LINKFLAGS_RELEASE}")
string(APPEND CMAKE_EXE_LINKER_FLAGS_DEBUG   "${LINKFLAGS_DEBUG}")
```

with the following code:

```
SET(CMAKE_C_FLAGS "${ARCHFLAGS} ${CCFLAGS} ${CFLAGS}" CACHE INTERNAL "c compiler flags")
SET(CMAKE_C_FLAGS_RELEASE   "${CCFLAGS_RELEASE}"      CACHE INTERNAL "c compiler flags release")
SET(CMAKE_C_FLAGS_DEBUG     "${CCFLAGS_DEBUG}"        CACHE INTERNAL "c compiler flags debug")

SET(CMAKE_CXX_FLAGS "${ARCHFLAGS} ${CCFLAGS} ${CXXFLAGS}" CACHE INTERNAL "cxx compiler flags")
SET(CMAKE_CXX_FLAGS_RELEASE "${CCFLAGS_RELEASE}"          CACHE INTERNAL "cxx compiler flags release")
SET(CMAKE_CXX_FLAGS_DEBUG   "${CCFLAGS_DEBUG}"            CACHE INTERNAL "cxx compiler flags debug")

SET(CMAKE_ASM_FLAGS "${ARCHFLAGS} ${ASFLAGS}" CACHE INTERNAL "asm compiler flags")
SET(CMAKE_ASM_FLAGS_RELEASE "${ASFLAGS_RELEASE}" CACHE INTERNAL "asm compiler flags release")
SET(CMAKE_ASM_FLAGS_DEBUG   "${ASFLAGS_DEBUG}" CACHE INTERNAL "asm compiler flags debug")

SET(CMAKE_EXE_LINKER_FLAGS  "${ARCHFLAGS} ${LINKFLAGS}" CACHE INTERNAL "linker flags")
SET(CMAKE_EXE_LINKER_FLAGS_RELEASE "${LINKFLAGS_RELEASE}" CACHE INTERNAL "linker flags release")
SET(CMAKE_EXE_LINKER_FLAGS_DEBUG   "${LINKFLAGS_DEBUG}" CACHE INTERNAL "linker flags debug")
```

salkinium, 2019-08-21:
That sounds reasonable, I'll check it out next week after holidays.

salkinium, 2021-05-08:
The CMake generator has been completely replaced making this probably obsolete.

closed 2021-05-08 by salkinium