modm-io/modm #271: enable semihosting

W-M-D, 2019-08-28:
Hello i was wondering how i could enable semi hosting with modm ?

salkinium, 2019-08-28:
With the [`modm:platform:itm` module](https://modm.io/reference/module/modm-platform-itm/), but be aware of its limitations!

salkinium, 2019-08-28:
There's also [an example here](https://github.com/modm-io/modm/blob/develop/examples/nucleo_f103rb/itm/main.cpp).

salkinium, 2019-08-28:
Technically this only provides logging, true semihosting (with  forwarding of libc functions for filesystem access etc) isn't implemented.

W-M-D, 2019-08-29:
Wow thanks for the quick response!

W-M-D, 2019-08-29:
Hm it seems that the module isn't available for the stm32f071x8t. Is there any way i can enable it ? I know it exists since i have used it with c before.

salkinium, 2019-08-29:
Oh, then there's no support for that in modm. 😧 The F0 is using a Cortex-M0, which is lacking the ITM for asynchronous logging, so the fallback would be to use semihosting IO for synchronous logging, but that's not implemented.

W-M-D, 2019-08-29:
ooofff

salkinium, 2019-08-29:
Booooooo! Using the semihosting IO solution means stopping the MCU during the data transfer (incl. interrupts), and that wasn't acceptable for most of our applications. So I prioritized the SWO async implementation.

W-M-D, 2019-08-29:
Oh that might be the reason for a lot of my issues...

salkinium, 2019-10-12:
I'm closing this, since you've found another solution.

closed 2019-10-12 by salkinium