modm-io/modm #276: ITM on Nucleo-F103RB broken with commit f69245a77

luxarf, 2019-09-06:
I'm using the ITM on a Nucleo-F103RB to log test results. To get the newest features I recently updated the modm submodule in my project and itm stopped working.

With a little bit of trial and error, I was able to pin down the problem to be emerging with commit f69245a77.

Steps to reproduce:
1. connect Nucleo-F103RB
2. cd modm/examples/nucleo_f103rb/itm
3. git checkout a05b31f64
4. lbuild build && scons program -j8 && scons itm fcpu=64000000
-> produces the expected output (loop: 1, loop: 2, etc.)

5. git checkout f69245a77
6. lbuild build && scons program -j8 && scons itm fcpu=64000000
-> no output

I am running Debian 9 with KDE, shell is zsh 5.3.1, terminal emulater is konsole 16.12.0.

So far I have not been able to find out whether it is the environment I am running or the changes introduced in f69245a77, since these apparently deal with how openocd is receiving it's config files.

Do you need more information? How can I help?

On  a side note: 
(I don't know, whether an issue is really the right place for it, but gonna say it anyway)
I am really loving the project, it makes working with these devices a breeze. Thank you for that.

salkinium, 2019-09-06:
Hm, that's probably because I didn't refactor correctly and missed to [`env.Listify` these openocd configs in the ITM SCons tool](https://github.com/modm-io/modm/blob/develop/tools/build_script_generator/scons/site_tools/log_itm.py#L23-L25).

> How can I help?

Can you wrap these into `env.Listify` and check if that fixes it? I'll see if I can find my Nucleo-F103 in the meantime.

salkinium, 2019-09-06:
> I'll see if I can find my Nucleo-F103 in the meantime.

I can reproduce the bug and also have a fix. Lemme add a PR.

luxarf, 2019-09-06:
That totally solves it.
Checked on commit f69245a and the current head of develop.
Thank you for your incredibly fast respose time.

salkinium, 2019-09-06:
> Thank you for your incredibly fast respose time.

The benefits of [push notifications](https://mxcl.dev/canopy/) and currently being awake 😉.

salkinium, 2019-09-06:
Also your bug report was super detailed and just seeing the commit made me understand the issue immediately.

luxarf, 2019-09-06:
> Also your bug report was super detailed and just seeing the commit made me understand the issue immediately.

Thank you, I am happy to hear that.

PR #277 (2019-09-06): [scons] Fix missing env.Listify for ITM tool
Fixes #276.

cc @luxarf
closed 2019-09-06 by salkinium