modm-io/modm #280: Getting Started guide is confusing

WasabiFan, 2019-09-09:
Hi!

First, I just wanted to say I quite like the look of the modm API and look forward to getting it running. I've been working through testing my scenarios and wanted to share some initial impressions. Once I know more about the library I might be able to come back and open a PR for this.

For now, here's a walk-through of my experience with the onboarding documentation and some thoughts on how it would make more sense _to me_. **It might come across negative. If so, I apologize; that isn't my intent.** This is intended to walk through my mindset and might let some confusion through. Either way, I really am just trying to communicate my experience as clearly as possible.

---

Here's what I see when I open the guide, for reference:

[image]

When I open the Getting Started page, I see two visible headings:  `TL;DR` and `Quickstart`. Both of these sections contain only commands that it tells you can be run, getting pretty deep into usage past just setup. The thing is... none of these actually tell you where you run them or how they are supposed to be used, and it turns out that they only work after you have cloned the modm source repository and put it in the right spot. **This was confusing point 1: the guide starts with commands without any context for where you run them or what you need beforehand.**

Now, assuming I figure out that I need the source _and_ figure out where it should be put, I need to create my project. What files are required for that? As it turns out, it's a `main.cpp` and a `project.xml`. But the only way to know that is to _scroll down_ beyond two more headings (ignoring the library code examples, which don't have to manually include modm and therefore are structured differently) and find the on-page example. By this time, you've scrolled through the whole guide and hopefully realized that there's a `Your own project` heading, which is _actually where you need to start_: `TL;DR` and `Quickstart` both skip over the crucial steps of having modm, source and project files on disk first. **This is confusing point 2: If I need to create/clone files, I need to be told that up-front and in-order.** `TL;DR` and `Quickstart` are both red herrings which distract from the things you need to do.

Now, the `Your own project` section has good info. But it has four headings (`Using a Board Support Package`, `Discovering modm`, `Custom Configuration`, and `Generate and Compile`) which don't seem to have appropriate grouping. What I _think_ it's trying to do is present two options: either use an existing BSP for a known board, or make a custom configuration. But why is `Discovering modm` in the middle? And if a BSP and a custom configuration are mutually exclusive, why are they listed as two peer headings in order? Additionally, `Discovering modm` has a lot of content and it isn't clear why I need to care as an introductory user. **This is confusing point 3: if I'm being presented with mutually exclusive options, I need to be told, and they should be grouped as such. The major steps and order in which they should be done must be clear.**

---

With all that being said, here's my suggestion for a restructuring. Start by moving `TL;DR` to the bottom, and moving any useful `Quickstart` content into what's now `Your own project`. Delete `Quickstart`. Move the "examples" section to the bottom, above `TL;DR`. While you're at it, maybe rename `TL;DR` to `Quickstart` because that seems like a better name anyway. The current `Your own project` is now the first section, and should tell you how to a) set up a project directory, b) get the right files in place, and c) run the commands to build. Then you can have a further section/subsection for running it on the device and debugging.

Over all, my suggested structure is:
- **Your first project**
  - **Setting up a project** (renamed from `Your own project`)
    - Create a directory, clone modm into it, create a subdirectory for the project files, add a default `project.xml` and `main.cpp`, and build.
    - The area that gives you default project file contents can link to a separate page or section below with info on _not_ using a BSP. The default should be to use one.
  - **Loading and debugging**
- **Library examples** (links to examples as it is now)
- **Quickstart**
  - Show the directory structure and the commands to run with a few sentences between each.

Again, I am nowhere near an expert, so some of this might be wrong/confused/shallow. I'd be happy to talk over ideas and all of this should be taken with a grain of salt; it's only impressions.

Some additional notes on the experience:
- The nav on the left of the page lists "Getting started" before "Installation", which seems backwards.
- Neither the installation page nor anywhere obvious on the getting started page makes it clear that the way you use modm in your project is by including its source. And further, none explain that the source isn't used "in situ", but rather a build command makes copies of it after evaluating templates in yet another directory. So if I open these pages, I'm left confused wondering where I'm supposed to get the library from. It would be helpful if there were a paragraph explaining how you get the library, how it's integrated into your own code's build, and how you actually use it.
- The header formatting on that page is hard to differentiate between `h1`, `h2`, and `h3`.

Thanks! I'd be happy to talk through any of this.

salkinium, 2019-09-09:
Thanks a lot for this feedback, it's very difficult to judge the effectiveness of the docs for me, since my head is so full of implicit design knowledge. I completely agree with your critique. The reason why TL;DR and Quickstart feel so out of place is because it's actually copied from the `examples/README.md` and thus from a different context. I think the guide tries too hard to describe customization points, that should be separated. I'll rework this on the weekend.

blake5634, 2020-08-09:
Related point to this:   one of the first steps is to do "lbuild build" but this returns
 lbuild build 
------------------------------------------------
ERROR: Configuration(.): no repositories loaded!
Hint: Add a path to one or more repositories, either in your configuration files:

    <repositories>
      <repository>
        <path>path/to/repo.lb</path>
      </repository>
    </repositories>

Or via the command line:

    lbuild -r path/to/repo.lb discover
------------------------------------------------------------
The suggested command line (just above) doesn't work either. 
(Ubuntu 18.04)

salkinium, 2020-08-09:
`lbuild build` is meant to be executed in an example, but the getting started guide is bad. I'll prioritize rewriting it, but I'm very busy with work and uni until the end of the year.

blake5634, 2020-08-09:
Thanks - this issue thread helped me break through the impass.
BH

blake5634, 2020-08-10:
Don't know if you have any spare cycles for this but in case you do, I'm
trying to get a basic setup going on Ubuntu 18.04 in a Python3 VE.
Finally modified search path as

env.Append(toolpath=[r"/home/blake/stm32/modm/tools/build_script_generator/scons/site_tools",

r"/home/blake/python_envs/stm/lib/python3.6/site-packages",])

to eliminate a lot of errors, but even though it finds one thing in
site-packages, I cant find gitdb (which IS installed)

scons: Reading SConscript files ...
 File
"/home/blake/python_envs/stm/lib/python3.6/site-packages/git/__init__.py",
line 27

   raise ImportError("'gitdb' could not be found in your PYTHONPATH") from
e

                                                                         ^

SyntaxError: invalid syntax

seems like gitdb (and similar deps) might be the wrong version???

I'm about to give up - the rest of my team seems to be going strong with
Windows.

Blake

salkinium, 2020-08-10:
> env.Append(toolpath=[r"/home/blake/stm32/modm/tools/build_script_generator/scons/site_tools"

How did you get there? What are you trying to do? Is it a custom project or an example? What does your `project.xml` look like?

You cannot use our SCons build tools in their raw state, they must first be processed by `lbuild build` to specialize to the selected target (the generated tools are *very* different for AVR and STM32). The generated tools are located in `modm/scons/site_tools` which is typically placed in the same folder as your `project.xml`.

Ubuntu 18.04 still has Python2 as default, so SCons will use the Python2 interpreter, unless you're using SCons 4.0 (Python 3 only) or explicitly call scons from Python3: `alias scons="/usr/bin/env python3 $(which scons)"`
I'm not sure about how a Python3 virtualenv would interact with SCons, I would expect virtualenv to be transparent to SCons though, as it should be to any Python tool.

salkinium, 2020-08-10:
> I'm about to give up - the rest of my team seems to be going strong with
Windows.

Ah wait, you're from https://github.com/uw-advanced-robotics right?
So I suspect that the team can compile the same code on Windows, but you cannot compile this on Ubuntu?

What are the errors? Which version of modm are you using? The ARUW modm fork which is 216 commits behind? https://github.com/uw-advanced-robotics/modm

WasabiFan, 2020-08-10:
I just checked in with others on my team and it sounds like this is a separate project which is using modm by our recommendation. I'll have them follow up privately and see if they can help resolve the setup issue. I don't think we've encountered this particular error, but as you said, we've internally pinned on an older modm build and won't see more recent changes!

The above errors were presumably from stock modm; they won't be using our fork, build system or setup config. We do indeed build on Windows, macOS and Linux but most of our protocols are probably not applicable to upstream or other projects.

Sorry for the confusion! I can appreciate the difficulty in taking issue reports on a rather large OSS project :slightly_smiling_face:

blake5634, 2020-08-10:
Thanks for getting back so quickly all.      This is for the World
Ventilator project.   I'll sync with them on which version.   But
I had to add those paths in order to make it find things like the tools.
Honestly it's ironic to work so hard to help a tool that's supposed to
automate a build.

lbuild build was done and doesn't fix it.
cat project.xml
<library>
 <repositories>
   <repository>
     <path>../modm/repo.lb</path>
   </repository>
 </repositories>
 <options>
   <option name="modm:build:build.path">./build</option>
   <option name="modm:target">stm32f070rbt6</option>
   <option name="modm:build:scons:include_sconstruct">False</option>
 </options>
 <modules>
   <module>modm:build:scons</module>
   <module>modm:architecture:clock</module>
   <module>modm:platform:clock</module>
   <module>modm:platform:core</module>
   <module>modm:platform:gpio</module>
   <module>modm:platform:heap</module>
   <module>modm:platform:adc</module>
   
 </modules>
</library>

Blake

WasabiFan, 2020-08-10:
@blake5634 I think you might have more luck if you open a separate issue for this -- the thread we're currently in is focused on the text and formatting of the getting started guide rather than troubleshooting a particular error.

salkinium, 2020-08-10:
> But I had to add those paths in order to make it find things like the tools.

Ok, so a very important information is right there: You're using a custom SConstruct: `<option name="modm:build:scons:include_sconstruct">False</option>`
This means there should be a SConstruct checked into the repo next to `project.xml`.

So what does the custom SConstruct look like? (It should reside in the folder your calling `scons` from.) Why are you using it? What other SCons tooling are you using?

blake5634, 2020-08-10:
Thanks,
    I just had to make mods to the original SConstruct (next to
project.xml) because it was failing to find all the tools and stuff without
those mods.
Blake

salkinium, 2021-03-29:
It's been a long time but I found the motivation to rewrite the guides with your feedback in mind. You can see the result in #592.

closed 2021-03-30 by salkinium