modm-io/modm #282: STM32F7 does not build

rleh, 2019-09-09:
While building doxygen documentation for many devices I discovered the following error, which occurs on all STM32F7* devices.
```
ERROR: In Module 'modm:freertos': File or directory not found!
In file 'modm/ext/aws/module.lb:58' in function 'build':
    env.copy("{}/port.c".format(path), "freertos/port.c")

Source path 'modm/ext/aws/freertos/FreeRTOS/Source/portable/GCC/ARM_CM7/port.c' does not exist!
Hint: Use 'localpath(path)' or 'repopath(path)' for relative paths.
```

salkinium, 2019-09-09:
Note to self: The FreeRTOS module is choosing the wrong port, it probably should choose the CM4F port. This clearly wasn't tested in any examples.

rleh, 2019-09-09:
Looks like some targets build:
* stm32f745zgt
* stm32f746zgy
* stm32f756zgy
* stm32f767zit
* stm32f779nih

rleh, 2019-09-09:
I tested this list of STM32F7 devices:
* stm32f722zet
* stm32f723zet
* stm32f730z8t
* stm32f732zet
* stm32f733zet
* stm32f745zgt
* stm32f746zgy
* stm32f750z8t
* stm32f756zgy
* stm32f765zit
* stm32f767zit
* stm32f768aiy
* stm32f769nih
* stm32f777zit
* stm32f778aiy
* stm32f779nih

salkinium, 2019-09-09:
This line should probably also take into consideration the M7FD (double-precision FPU) devices: https://github.com/modm-io/modm/blob/develop/ext/aws/module.lb#L47

salkinium, 2019-09-09:
Whatever the bug, the core type needs to map entirely to these folders: https://github.com/modm-ext/freertos-partial/tree/master/FreeRTOS/Source/portable/GCC