modm-io/modm #283: Device cache not recomputing after modifying repo.lb

rleh, 2019-09-09:
The modm device cache (file `modm/ext/modm-devices.cache`) is not updated after a change to `modm/repo.lb`.

salkinium, 2019-09-09:
The cache is only recomputed when it has been deleted or if the modm (or modm-devices) repository git commit SHA changed (ie. you committed something). Uncommitted changes are not tracked since that would development really annoying.

salkinium, 2019-09-09:
What changes are you making to repo.lb? I could see this detection useful for when you change the filters, but really there are more dependencies to track than just repo.lb.

rleh, 2019-09-09:
From my point of view the device cache should be recomputed as soon as any input changes. That means modm-devices as well as all scripts involved in the computation.
Otherwise the cache is nonsense...

salkinium, 2019-09-09:
The cache needs to go into modm-devices and have an actual API with invalidation. Right now it's just hacked together.

salkinium, 2026-10-01:
Oh, this was fixed! There's no cache anymore, modm-devices has a json with the mapping directly.

closed 2026-10-01 by salkinium