modm-io/modm #284: arm-none-eabi-gcc 9.2.0 VFP linker error

rleh, 2019-09-09:
When compiling a STM32F429 example on my Linux/Fedora 30 machine with GCC 9.2 I get the following error:
```
.../arm-none-eabi/bin/ld: error: /[...]/release/blink.elf uses VFP register arguments,
.../arm-none-eabi/lib/libm.a(lib_a-s_round.o) does not
```

<details>
<summary>Complier flags: "-mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb"</summary>
```
arm-none-eabi-g++ -o [...]/blink/release/main.o -c -fno-exceptions -fno-rtti -fno-threadsafe-statics -fno-unwind-tables -fstrict-enums -fuse-cxa-atexit -std=c++17 -Woverloaded-virtual -fdata-sections -ffunction-sections -finline-limit=10000 -fshort-wchar -fsingle-precision-constant -funsigned-bitfields -funsigned-char -fwrapv -g3 -gdwarf -W -Wall -Wdouble-promotion -Wduplicated-cond -Werror=format -Werror=maybe-uninitialized -Werror=overflow -Werror=sign-compare -Wextra -Wlogical-op -Wpointer-arith -Wundef -Os -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -Imodm/ext -Imodm/ext/cmsis/core -Imodm/ext/cmsis/device -Imodm/src -I. main.cpp
```
</details>

<details>
<summary>Compiler: arm-none-eabi-gcc 9.2.0 on Fedora 30 (9.2.0-1.fc30)</summary>

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Fedora 9.2.0-1.fc30) 9.2.0
```

```bash
$ arm-none-eabi-gcc -print-multi-lib
.;
arm/v5te/softfp;@marm@march=armv5te+fp@mfloat-abi=softfp
arm/v5te/hard;@marm@march=armv5te+fp@mfloat-abi=hard
thumb/nofp;@mthumb@mfloat-abi=soft
thumb/v7/nofp;@mthumb@march=armv7@mfloat-abi=soft
thumb/v7+fp/softfp;@mthumb@march=armv7+fp@mfloat-abi=softfp
thumb/v7+fp/hard;@mthumb@march=armv7+fp@mfloat-abi=hard
thumb/v6-m/nofp;@mthumb@march=armv6s-m@mfloat-abi=soft
thumb/v7-m/nofp;@mthumb@march=armv7-m@mfloat-abi=soft
thumb/v7e-m/nofp;@mthumb@march=armv7e-m@mfloat-abi=soft
thumb/v7e-m+fp/softfp;@mthumb@march=armv7e-m+fp@mfloat-abi=softfp
thumb/v7e-m+fp/hard;@mthumb@march=armv7e-m+fp@mfloat-abi=hard
thumb/v7e-m+dp/softfp;@mthumb@march=armv7e-m+fp.dp@mfloat-abi=softfp
thumb/v7e-m+dp/hard;@mthumb@march=armv7e-m+fp.dp@mfloat-abi=hard
thumb/v8-m.base/nofp;@mthumb@march=armv8-m.base@mfloat-abi=soft
thumb/v8-m.main/nofp;@mthumb@march=armv8-m.main@mfloat-abi=soft
thumb/v8-m.main+fp/softfp;@mthumb@march=armv8-m.main+fp@mfloat-abi=softfp
thumb/v8-m.main+fp/hard;@mthumb@march=armv8-m.main+fp@mfloat-abi=hard
thumb/v8-m.main+dp/softfp;@mthumb@march=armv8-m.main+fp.dp@mfloat-abi=softfp
thumb/v8-m.main+dp/hard;@mthumb@march=armv8-m.main+fp.dp@mfloat-abi=hard
```
</details>

What is the issue here?

rleh, 2019-09-10:
Looks like the linker uses
`/usr/arm-none-eabi/lib/libc_nano.a` and `/usr/arm-none-eabi/lib/libm.a`
instead of
`/usr/arm-none-eabi/lib/hard/libc_nano.a` and `/usr/arm-none-eabi/lib/hard/libm.a`.

If I overwrite the libraries in `lib/` with the libraries from `lib/hard/` linking works fine.

Is this a GCC, a Fedora packaging or a modm issue?

salkinium, 2019-09-10:
`modm:build` only specifies [the `modm:build:archflags`](https://github.com/modm-io/modm/blob/develop/tools/build_script_generator/common.py#L278-L295) once and they are then added to all relevant tools. That's all modm can do.

Then the GCC compiler uses [internal configuration files called `spec files`](https://gcc.gnu.org/onlinedocs/gcc/Spec-Files.html), which translate the compile flags into more specific commands, for example, which libc/libm to include for what flags.

There *may* be an issue there (check `arm-none-eabi-gcc -dumpspecs` output for `*multilib` entries), but then again every time I sought to blame the compiler I was wrong…

salkinium, 2019-09-10:
On the other hand, `thumb/v7e-m+fp/hard;@mthumb@march=armv7e-m+fp@mfloat-abi=hard` is what should be used for the F4 series, so why is it not?

rleh, 2019-09-10:
`arm-none-eabi-gcc -dumpspecs`

```
*multilib:

. !marm !mthumb !march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp !mfloat-abi=soft !mfloat-abi=softfp !mfloat-abi=hard;arm/v5te/softfp marm !mthumb march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp !mfloat-abi=soft mfloat-abi=softfp !mfloat-abi=hard;arm/v5te/hard marm !mthumb march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp !mfloat-abi=soft !mfloat-abi=softfp mfloat-abi=hard;thumb/nofp !marm mthumb !march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp mfloat-abi=soft !mfloat-abi=softfp !mfloat-abi=hard;thumb/v7/nofp !marm mthumb !march=armv5te+fp march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp mfloat-abi=soft !mfloat-abi=softfp !mfloat-abi=hard;thumb/v7+fp/softfp !marm mthumb !march=armv5te+fp !march=armv7 march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp !mfloat-abi=soft mfloat-abi=softfp !mfloat-abi=hard;thumb/v7+fp/hard !marm mthumb !march=armv5te+fp !march=armv7 march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp !mfloat-abi=soft !mfloat-abi=softfp mfloat-abi=hard;thumb/v6-m/nofp !marm mthumb !march=armv5te+fp !march=armv7 !march=armv7+fp march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp mfloat-abi=soft !mfloat-abi=softfp !mfloat-abi=hard;thumb/v7-m/nofp !marm mthumb !march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m march=armv7-m !march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp mfloat-abi=soft !mfloat-abi=softfp !mfloat-abi=hard;thumb/v7e-m/nofp !marm mthumb !march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp mfloat-abi=soft !mfloat-abi=softfp !mfloat-abi=hard;thumb/v7e-m+fp/softfp !marm mthumb !march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp !mfloat-abi=soft mfloat-abi=softfp !mfloat-abi=hard;thumb/v7e-m+fp/hard !marm mthumb !march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp !mfloat-abi=soft !mfloat-abi=softfp mfloat-abi=hard;thumb/v7e-m+dp/softfp !marm mthumb !march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp !mfloat-abi=soft mfloat-abi=softfp !mfloat-abi=hard;thumb/v7e-m+dp/hard !marm mthumb !march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp !mfloat-abi=soft !mfloat-abi=softfp mfloat-abi=hard;thumb/v8-m.base/nofp !marm mthumb !march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp march=armv8-m.base !march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp mfloat-abi=soft !mfloat-abi=softfp !mfloat-abi=hard;thumb/v8-m.main/nofp !marm mthumb !march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base march=armv8-m.main !march=armv8-m.main+fp !march=armv8-m.main+fp.dp mfloat-abi=soft !mfloat-abi=softfp !mfloat-abi=hard;thumb/v8-m.main+fp/softfp !marm mthumb !march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main march=armv8-m.main+fp !march=armv8-m.main+fp.dp !mfloat-abi=soft mfloat-abi=softfp !mfloat-abi=hard;thumb/v8-m.main+fp/hard !marm mthumb !march=armv5te+fp !march=armv7 !march=armv7+fp !march=armv6s-m !march=armv7-m !march=armv7e-m !march=armv7e-m+fp !march=armv7e-m+fp.dp !march=armv8-m.base !march=armv8-m.main march=armv8-m.main+fp !march=armv8-m.main+fp.d [... cut ...] nitize=thread}}    %{%:sanitize(undefined):%{static-libubsan:-Bstatic} -lubsan %{static-libubsan:-Bdynamic} %{static-libubsan|static:%:include(libsanitizer.spec)%(link_libubsan)}}    %{%:sanitize(leak):%{static-liblsan:-Bstatic} -llsan %{static-liblsan:-Bdynamic} %{static-liblsan|static:%:include(libsanitizer.spec)%(link_liblsan)}}}}}     %{!nostdlib:%{!r:%{!nodefaultlibs:%(link_ssp) %(link_gcc_c_sequence)}}}    %{!nostdlib:%{!r:%{!nostartfiles:%E}}} %{T*}  
%(post_link) }}}}}}
</pre>
</details>

salkinium, 2019-09-10:
Hm, now what?

<details>
<summary>$ diff specs.mine specs.yours</summary>

```diff
 $ diff specs.mine specs.yours
8,10c8,10
< %{gsplit-dwarf:
<        objcopy --extract-dwo 	 %{c:%{o*:%*}%{!o*:%b%O}}%{!c:%U%O} 	 %{c:%{o*:%:replace-extension(%{o*:%*} .dwo)}%{!o*:%b.dwo}}%{!c:%b.dwo}
<        objcopy --strip-dwo 	 %{c:%{o*:%*}%{!o*:%b%O}}%{!c:%U%O}     }
---
> %{gsplit-dwarf:
>        objcopy --extract-dwo     %{c:%{o*:%*}%{!o*:%b%O}}%{!c:%U%O}      %{c:%{o*:%:replace-extension(%{o*:%*} .dwo)}%{!o*:%b.dwo}}%{!c:%b.dwo}
>        objcopy --strip-dwo       %{c:%{o*:%*}%{!o*:%b%O}}%{!c:%U%O}     }
20c20
< %(subtarget_cpp_spec)					%{mfloat-abi=soft:%{mfloat-abi=hard:				%e-mfloat-abi=soft and -mfloat-abi=hard may not be used together}} %{mbig-endian:%{mlittle-endian:			%e-mbig-endian and -mlittle-endian may not be used together}}
---
> %(subtarget_cpp_spec)                                   %{mfloat-abi=soft:%{mfloat-abi=hard:                                            %e-mfloat-abi=soft and -mfloat-abi=hard may not be used together}} %{mbig-endian:%{mlittle-endian: %e-mbig-endian and -mlittle-endian may not be used together}}
29c29
< %{!Q:-quiet} %{nostdinc*} %{C} %{CC} %{v} %{I*&F*} %{P} %I %{MD:-MD %{!o:%b.d}%{o*:%.d%*}} %{MMD:-MMD %{!o:%b.d}%{o*:%.d%*}} %{M} %{MM} %{MF*} %{MG} %{MP} %{MQ*} %{MT*} %{!E:%{!M:%{!MM:%{!MT:%{!MQ:%{MD|MMD:%{o*:-MQ %*}}}}}}} %{remap} %{g3|ggdb3|gstabs3|gxcoff3|gvms3:-dD} %{!iplugindir*:%{fplugin*:%:find-plugindir()}} %{H} %C %{D*&U*&A*} %{i*} %Z %i %{E|M|MM:%W{o*}}
---
> %{!Q:-quiet} %{nostdinc*} %{C} %{CC} %{v} %@{I*&F*} %{P} %I %{MD:-MD %{!o:%b.d}%{o*:%.d%*}} %{MMD:-MMD %{!o:%b.d}%{o*:%.d%*}} %{M} %{MM} %{MF*} %{MG} %{MP} %{MQ*} %{MT*} %{!E:%{!M:%{!MM:%{!MT:%{!MQ:%{MD|MMD:%{o*:-MQ %*}}}}}}} %{remap} %{g3|ggdb3|gstabs3|gxcoff3|gvms3:-dD} %{!iplugindir*:%{fplugin*:%:find-plugindir()}} %{H} %C %{D*&U*&A*} %{i*} %Z %i %{E|M|MM:%W{o*}}
44c44
< --start-group %G %L --end-group
---
> --start-group %G %{!nolibc:%L} --end-group
53c53
< %{mbig-endian:-EB} %{mlittle-endian:-EL} %{static:-Bstatic} %{shared:-shared} %{symbolic:-Bsymbolic} -X %{mcpu=arm8|mcpu=arm810|mcpu=strongarm*|march=armv4|mcpu=fa526|mcpu=fa626:--fix-v4bx}%{!r:%{!mbe32:%:be8_linkopt(%{mlittle-endian:little}			       %{mbig-endian:big}			       %{mbe8:be8}			       %{march=*:arch %*})}}
---
> %{mbig-endian:-EB} %{mlittle-endian:-EL} %{static:-Bstatic} %{shared:-shared} %{symbolic:-Bsymbolic} -X %{mcpu=arm8|mcpu=arm810|mcpu=strongarm*|march=armv4|mcpu=fa526|mcpu=fa626:--fix-v4bx}%{!r:%{!mbe32:%:be8_linkopt(%{mlittle-endian:little}                          %{mbig-endian:big}                              %{mbe8:be8}                             %{march=*:arch %*})}}
71c71
< 8.3.1
---
> 9.2.0
83c83
< march=armv6-m march=armv6s-m;march=armv8-m.main+dsp march=armv8-m.main;march=armv8-m.main+dsp+fp march=armv8-m.main+fp;march=armv8-m.main+dsp+fp.dp march=armv8-m.main+fp.dp;march=armv7e-m+fpv5 march=armv7e-m+fp;march=armv7-r march=armv7;march=armv7-r+idiv march=armv7;march=armv7-r+fp march=armv7+fp;march=armv7-r+fp+idiv march=armv7+fp;march=armv5e+fp march=armv5te+fp;march=armv5tej+fp march=armv5te+fp;march=armv6+fp march=armv5te+fp;march=armv6j+fp march=armv5te+fp;march=armv6k+fp march=armv5te+fp;march=armv6z+fp march=armv5te+fp;march=armv6kz+fp march=armv5te+fp;march=armv6zk+fp march=armv5te+fp;march=armv6t2+fp march=armv5te+fp;march=iwmmxt+fp march=armv5te+fp;march=iwmmxt2+fp march=armv5te+fp;march=armv8-r march=armv7;march=armv8-r+crc march=armv7;march=armv8-r+simd march=armv7+fp;march=armv8-r+crc+simd march=armv7+fp;march=armv8-r+crypto march=armv7+fp;march=armv8-r+crc+crypto march=armv7+fp;march=armv7-a march=armv7;march=armv7-a+fp march=armv7+fp;march=armv7-a+vfpv3 march=armv7+fp;march=armv7-a+vfpv3-d16-fp16 march=armv7+fp;march=armv7-a+vfpv3-fp16 march=armv7+fp;march=armv7-a+vfpv4-d16 march=armv7+fp;march=armv7-a+vfpv4 march=armv7+fp;march=armv7-a+simd march=armv7+fp;march=armv7-a+neon-fp16 march=armv7+fp;march=armv7-a+neon-vfpv4 march=armv7+fp;march=armv7ve march=armv7;march=armv7ve+vfpv3-d16 march=armv7+fp;march=armv7ve+vfpv3 march=armv7+fp;march=armv7ve+vfpv3-d16-fp16 march=armv7+fp;march=armv7ve+vfpv3-fp16 march=armv7+fp;march=armv7ve+fp march=armv7+fp;march=armv7ve+vfpv4 march=armv7+fp;march=armv7ve+neon march=armv7+fp;march=armv7ve+neon-fp16 march=armv7+fp;march=armv7ve+simd march=armv7+fp;march=armv8-a march=armv7;march=armv8-a+crc march=armv7;march=armv8-a+crc+simd march=armv7+fp;march=armv8-a+simd march=armv7+fp;march=armv8-a+crc+simd march=armv7+fp;march=armv8-a+simd+crypto march=armv7+fp;march=armv8-a+crc+simd+crypto march=armv7+fp;march=armv8-a+crypto march=armv7+fp;march=armv8-a+crc+crypto march=armv7+fp;march=armv8.1-a march=armv7;march=armv8.1-a+simd march=armv7+fp;march=armv8.1-a+simd+crypto march=armv7+fp;march=armv8.1-a+crypto march=armv7+fp;march=armv8.2-a march=armv7;march=armv8.3-a march=armv7;march=armv8.2-a+simd march=armv7+fp;march=armv8.3-a+simd march=armv7+fp;march=armv8.2-a+simd+fp16 march=armv7+fp;march=armv8.3-a+simd+fp16 march=armv7+fp;march=armv8.2-a+fp16 march=armv7+fp;march=armv8.3-a+fp16 march=armv7+fp;march=armv8.2-a+simd+fp16+fp16fml march=armv7+fp;march=armv8.3-a+simd+fp16+fp16fml march=armv7+fp;march=armv8.2-a+fp16+fp16fml march=armv7+fp;march=armv8.3-a+fp16+fp16fml march=armv7+fp;march=armv8.2-a+simd+fp16fml march=armv7+fp;march=armv8.3-a+simd+fp16fml march=armv7+fp;march=armv8.2-a+fp16fml march=armv7+fp;march=armv8.3-a+fp16fml march=armv7+fp;march=armv8.2-a+simd+fp16+fp16fml+cr [... cut ...] atic with -fsanitize=thread}}    %{%:sanitize(undefined):%{static-libubsan:-Bstatic} -lubsan %{static-libubsan:-Bdynamic} %{static-libubsan|static:%:include(libsanitizer.spec)%(link_libubsan)}}    %{%:sanitize(leak):%{static-liblsan:-Bstatic} -llsan %{static-liblsan:-Bdynamic} %{static-liblsan|static:%:include(libsanitizer.spec)%(link_liblsan)}}}}}     %{!nostdlib:%{!r:%{!nodefaultlibs:%(link_ssp) %(link_gcc_c_sequence)}}}    %{!nostdlib:%{!r:%{!nostartfiles:%E}}} %{T*}
151d150
<
```

</details>

salkinium, 2019-09-10:
Also, 🤓🤓technically🤓🤓, your toolchain isn't supported by modm. #RuleLawyering

> Install the ARM toochain by downloading the pre-built version for 64-bit Linux and adding its /bin directory to your path. Even though your distribution may ship their own ARM toolchain, we very strongly recommend using the official toolchain, since all of modm is tested with it.\

(from https://modm.io/guide/installation/#linux)

rleh, 2019-09-10:
Could someone else using the arm-none-eabi toolchain provided by their Linux distribution be kind enough to check if they get the same bug? @ekiwi @chris-durand @mhthies @daniel-k

salkinium, 2019-10-06:
Any new insight on this issue?

rleh, 2019-10-07:
Looks like Fedoras *arm-none-eabi-newlib* update from September 11 has fixed the issue.

closed 2019-10-07 by rleh