modm-io/modm #290: I2C EEPROM driver get stalled when writing

nesos, 2019-09-10:
Greetings,
I'm using the `i2cEeprom` module to read and write data to a FM24CL64 FRAM chip. The firmware is running on a STM32F042K6 chip.

Reading data from the FRAM works without any problems but writing stalls the whole chip (as I call the write method with `RF_CALL_BLOCKING`).

When sniffing on the I2C bus, the two address bytes got transmitted and afterwards the chips holds the clock down on ground and seems to be in an endless loop waiting for the transaction to finish.

I pinned  down the problem to the `modm::i2cEeprom::DataTransmissionAdapter::writing()` function in the i2c_eeprom.cpp file. The method only got called once for the address bytes but then never called again. Weirdly, when i change the `next` parameter to `OperationAfterWrite::Restart`, it will be called again, and the data will be put on the bus. 

Also if I don't use the driver but instead building custom transactions to communicate with the chip, the write operation works that way. So there is no hardware issue.

Does anyone have a clue what could cause the problem? I was digging into the code of the I2C Master but couldn't find anything obvious that could cause the problem.

Thanks in advance, Ben

salkinium, 2019-09-10:
Ugh, I2C is a shit show, the protocol sucks, the peripherals suck, the external chips suck, we've had nothing but trouble with that. 🤬 The transaction model is also annoying to use and inefficient \*writes on TODO list\*.

I was initialy thinking about clock stretching, but the FRAM chip doesn't need to do that since it finishes writing really fast. You're using the stm32-extended driver for I2C, which moved more protocol state into hardware. I didn't write that driver, and I suspect that it was never tested with a write-write operation 🥴 (cc @strongly-typed).

Is there a difference in behavior between using the hardware I2C and the bit-bang I2C? 
```cpp
modm::platform::BitBangI2cMaster<Scl, Sda> MyI2cMaster;
MyI2cMaster::connect<GpioA0::BitBang, GpioA1::BitBang>();
```

You can also enable some debug output for both drivers:
https://github.com/modm-io/modm/blob/develop/src/modm/platform/i2c/stm32-extended/i2c_master.cpp.in#L34
https://github.com/modm-io/modm/blob/develop/src/modm/platform/i2c/bitbang/bitbang_i2c_master_impl.hpp#L24

nesos, 2019-09-10:
Found the bug in the I2C Master ISR and fixed it. See pull request #291 for infos. Writing to the EEPROM works now!

closed 2019-09-10 by nesos

salkinium, 2019-09-10:
(You can close issues via the PR: https://help.github.com/en/articles/closing-issues-using-keywords)

PR #291 (2019-09-10): Fixed condition in I2C ISR for continued writes
Fixed the condition to wait for continued writes.
When a write operation followed by another write operation occurs, the ISR will falsely wait for a STOP condition, which will not be triggered as more data from the following write operation has to be transmitted.
This fix will wait for a TC interrupt if the next operation is a restart **or a write operation** and wait for a STOP interrupt otherwise

Fixes #290.