modm-io/modm #298: STM32F303x8 USART1 clock defaults to PCLK1

mikewolfram, 2019-10-02:
The clock for the USART1 on the Nucleo-F303K8 board defaults to PCLK1, which is divided by 2. The line

` static constexpr uint32_t Usart1 = Apb2;`

must be changed to

` static constexpr uint32_t Usart1 = Apb1;`

The reference manual tells it's connected to APB2, but the clock configuration register CFGR3 sets it to PCLK1 per default. See pages 162ff.

[image]

salkinium, 2019-10-02:
😱🥴

The board files are written by hand, which makes them error prone like this. It's on my long TODO list to automate this away.

Do you want to open a quick PR for this issue (and maybe check the remaining clocks too)?

mikewolfram, 2019-10-04:
Will look into it when I return from vacation.

mikewolfram, 2019-10-15:
Looking through the docs again I found out, that I2C1 default clock is HSI (8 MHz). I'll create a PR.

PR #304 (2019-10-17): Fixes clocks of USART1 and I2C1 on Nucleo F303K8 board.
According to data sheet USART1 clock defaults to PCLK1 and I2C1 clock to HSI.

Fixes #298.
closed 2019-10-17 by salkinium