modm-io/modm #300: CAN RX does not return data on STM32F4

WasabiFan, 2019-10-04:
I'm trying to interface with CAN motors on a board with an STM32F427. This board is known to work when using CubeMX init code.

Transmit works, but receive does not. I have tested RX by connecting to motor controllers that use CAN and also tried receiving from another board of the same type.

I use the code below to initialize CAN1 and CAN2:

```c++
CanFilter::setStartFilterBankForCan2(14);

// initialize CAN 1
Can1::connect<GpioD0::Rx, GpioD1::Tx>(Gpio::InputType::PullUp);
Can1::initialize<Board::SystemClock, 1000_kbps>(9);

// receive every message for CAN 1
CanFilter::setFilter(0, CanFilter::FIFO0, CanFilter::ExtendedIdentifier(0), CanFilter::ExtendedFilterMask(0));

// initialize CAN 2
Can2::connect<GpioB12::Rx, GpioB13::Tx>(Gpio::InputType::PullUp);
Can2::initialize<Board::SystemClock, 1000_kbps>(12);

// receive every message for CAN 2
CanFilter::setFilter(14, CanFilter::FIFO0, CanFilter::ExtendedIdentifier(0), CanFilter::ExtendedFilterMask(0));
```

Our main method is as follows:

```c++
int main()
{
    Board::initialize();
    while (1)
    {
        if (Can1::isMessageAvailable())
        {
            modm::can::Message message;
            Can1::getMessage(message);
            Board::Leds::toggle();
        }
        modm::delayMilliseconds(1);
    }
    return 0;
}
```

The LED never toggles and the ISRs `MODM_ISR(CAN1_RX0)` and `MODM_ISR(CAN1_RX1)` are never triggered, as far as I can tell.

What steps should I take to debug this? Is there something I'm missing?

salkinium, 2019-10-04:
There's an example with a loopback test: https://github.com/modm-io/modm/blob/develop/examples/stm32f4_discovery/can/main.cpp

If you can get that to work with your pins and then iterate yourself towards your setup, that could help.

salkinium, 2019-10-04:
What CAN transceivers are you using? Voltage range ok? Speed ok?

WasabiFan, 2019-10-05:
> There's an example with a loopback test: https://github.com/modm-io/modm/blob/develop/examples/stm32f4_discovery/can/main.cpp

Are you suggesting we tie CAN1 and CAN2 together for that loopback demo?

MatthewMArnold, 2019-10-05:
I am working with @WasabiFan on this project. Here is the datasheet for the CAN transceiver we are using: (http://ww1.microchip.com/downloads/en/DeviceDoc/20005167C.pdf). The potential difference between the high and low CAN lines we measured as ~2.7 volts.

salkinium, 2019-10-05:
Yes, put CAN1 and CAN2 on separate transceivers and connect those. Or use two boards with the same code.

WasabiFan, 2019-10-06:
We figured it out! The filter in the examples is set to `ExtendedIdentifier` and `ExtendedFilterMask`, whereas our devices were transmitting standard IDs. Where can you add that to the documentation? Maybe add a comment to the examples?

salkinium, 2019-10-06:
You can add this to the examples and also to the `CanFilter` class documentation.

WasabiFan, 2019-10-12:
Should it be a comment in every CAN example?

salkinium, 2019-10-12:
> Should it be a comment in every CAN example?

Sure, why not. Our examples are duplicated anyways, I'm aware of this issue, there was some [discussion about (not) having more generic examples some time ago](https://github.com/modm-io/modm/pull/73#issuecomment-416001467).

salkinium, 2019-11-08:
I'm closing this, since the original issue was resolved.

closed 2019-11-08 by salkinium