modm-io/modm #301: synchronize_docs.py script is broken

rleh, 2019-10-05:
```
$ ./tools/scripts/synchronize_docs.py 
Traceback (most recent call last):
  File "/home/user/.local/lib/python3.7/site-packages/lbuild/option.py", line 333, in as_enumeration
    return self._enumeration[self._obj_to_str(value)]
  File "/home/user/src/modm/repo.lb", line 152, in __getitem__
    value = dict.__getitem__(self, item)
KeyError: 'stm32f407vgt'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user/.local/lib/python3.7/site-packages/lbuild/option.py", line 58, in __set_value
    self._output = self._out(value)
  File "/home/user/.local/lib/python3.7/site-packages/lbuild/option.py", line 335, in as_enumeration
    raise TypeError("Input must be an element of the enumeration!")
TypeError: Input must be an element of the enumeration!

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user/.local/lib/python3.7/site-packages/lbuild/parser.py", line 150, in merge_repository_options
    option.value = value
  File "/home/user/.local/lib/python3.7/site-packages/lbuild/option.py", line 72, in value
    self._set_value(value)
  File "/home/user/.local/lib/python3.7/site-packages/lbuild/option.py", line 52, in _set_value
    self.__set_value(value)
  File "/home/user/.local/lib/python3.7/site-packages/lbuild/option.py", line 60, in __set_value
    raise le.LbuildOptionInputException(self, value, error)
lbuild.exception.LbuildOptionInputException: EnumerationOption(modm:target): input 'stm32f407vgt' is invalid! Input must be an element of the enumeration!

>> modm:target  [EnumerationOption]

Meta-HAL target device

Value: REQUIRED
Inputs: [at90can128-16ai, ...
```

salkinium, 2019-10-05:
You updated modm-devices to the latest version, instead of the modm submodule revision, it's currently not compatible with modm, due to https://github.com/modm-io/modm-devices/pull/29.

salkinium, 2019-10-05:
To preempt your protest: modm is not intended and not guaranteed to work with the HEAD versions of its submodules! We have no automated regression testing for many of the submodule features, thus this integration must occur manually.

rleh, 2019-10-05:
Oh sorry, I updated the submodule accidentally to HEAD during rebasing #287.

closed 2019-10-05 by rleh