modm-io/modm #312: Documentation: Highlight lbuild module names

rleh, 2019-11-03:
The module names (for project.xml) should be mentioned in the module documentation (https://modm.io) and in the Doxygen/Doxypress documents (https://docs.modm.io/).

salkinium, 2019-11-03:
Currently the module name is part of the title for the module docs, but not the API docs.
[image]

For Doxygen the module name is already encoded into the options, so it should be made explicit anyways.
[image]

How would you like to see the module name be emphasized?

rleh, 2019-11-03:
If someone doesn't know what the module name looks like, it's hard to guess that this strange beginning in the title is the module name.

I would suggest to add a line below the title like for the #include-file:

*Module name: `modm:ui:led`*

salkinium, 2019-11-03:
Ok, agreed. 

What about: `<module>modm:ui:led</module>` ?

rleh, 2019-11-03:
Exaktly :+1: 
I think the monospace/code-like formatting is important.

rleh, 2019-11-03:
Something similar would be nice inside doxygen API docs.
I guess this has to be added here:
https://github.com/modm-io/modm/blob/3eb25dbb18fc88328bf0e243ed352de774316135/tools/doc_generator/group.dox.in#L7-L9

PR #313 (2019-11-08): [docs] Explicitly declare the lbuild module in docs
Closes #312 

I originally wanted to use:

> lbuild module: `<module>name</module>`

but that was too much `module` for me. ;-P

Any better ideas? cc @rleh
closed 2019-11-08 by salkinium