modm-io/modm #315: Support for RTC

dirkbr, 2019-11-15:
It would be nice to have support for the internal RTC of the STM32F103 series.

salkinium, 2019-11-15:
What features would the implementation require? What is your use-case for using the RTC?

salkinium, 2019-11-15:
Ideally the API would implement the libc and libc++ interfaces for time and date, right? So the interface is kinda already defined, which is nice/.

dirkbr, 2019-11-15:
A libc and libc++ API would be perfect! One use-case is a mobile data logger, that doesn't have a permanent power supply.

salkinium, 2019-11-15:
Note that modm is not designed for doing anything fancy with power (neither sleeping,  nor purely interrupt based operations, nor non-volatile storage, nor partial power operations), so check if you can meet your power budget *before* relying on modm too much.

closed 2021-09-29 by salkinium